noble/node-bluetooth-hci-socket #3: Following issue 233 from Noble : compatibility with Intel adapter

pierreduf, 2015-08-16:
Following issue https://github.com/sandeepmistry/noble/issues/233

After having read some part of the linux BT drivers, it seems obvious that the Intel adapters work a bit differently from other one. To have them work they ;

1) Put the adapter in "developer mode" to get the firmware
2) Patch the firmware and reinject it
3) Reset the adapter (it seems to be a special command to reset as well).

https://github.com/torvalds/linux/blob/master/drivers/bluetooth/btusb.c

I'm gonna try to reproduce the same thing adapting your work on this repo (and thx to the "usb" thing in node.js) but I'm afraid it's a bit beyond my field so help would be welcome :)

Anyway, I guess Intel adapters represent a decent chunk of the market and everyone will be happy to use them :) !

Keep you posted.

pierreduf, 2015-08-17:
Initialiazing a noble object I get the following debug messages:

>   hci-usb HCI event: 0e0402030c00 +20ms
  hci onSocketData: 040e0402030c00 +0ms
  hci   event type = 4 +0ms
  hci   sub event type = 14 +0ms
  hci           cmd = 3075 +0ms
  hci           status = 0 +0ms
  hci           result =  +0ms

sandeepmistry, 2015-08-17:
@pierreduf interesting, here's by understanding of the break down:

```
0e     // EVT_CMD_COMPLETE
04     // length 
01     // ncmd
030c   // command - 0x0C03 = 3075 => reset
00     // status - 00 => success
```

See: http://git.kernel.org/cgit/bluetooth/bluez.git/tree/lib/hci.h#n1849

So it looks like the ``ncmd`` field value doesn't need to be an exact match to ``1``, it's just the command number. If you verify this works with your Intel adapter. I'd be happy to merge a pull request or make the change ...

Btw, what phone are your using (hardware, and OS)?

pierreduf, 2015-08-17:
It's a Huawei P8 Lite that has BLE (I can use it to pilot my rolling spider :))

Do you think I should see it using "your" scanning ? Because if I sjould, it's not the case so there's another pbm beyond the "wrong" check of the reset :(

sandeepmistry, 2015-08-17:
@pierreduf not sure if your phone is advertising, if it's running Android 5.0 or above it does have built-in peripheral API's.

I expect noble to only report BLE advertisements anyways.

The better question is does noble's [advertisement-discovery.js](https://github.com/sandeepmistry/noble/blob/master/examples/advertisement-discovery.js) example detect your Rolling Spider?

pierreduf, 2015-08-17:
Nope it doesn't work :(

I tried with your source, removing the "onStateChange" function and adding at the end:

noble.state = "poweredOn"
noble.startScanning();

The output is:

  hci-usb using USB VID = 32903, PID = 2602 +0ms
  hci-usb reset +20ms
  hci-usb write: 01030c00 +0ms
  hci set scan enabled - writing: 010c20020101 +0ms
  hci-usb write: 010c20020101 +20ms
  hci-usb HCI event: 0e0402030c00 +0ms
  hci onSocketData: 040e0402030c00 +10ms
  hci   event type = 4 +0ms
  hci   sub event type = 14 +0ms
  hci           cmd = 3075 +0ms
  hci           status = 0 +0ms
  hci           result =  +10ms
  hci-usb HCI event: 0e04020c2000 +20ms
  hci onSocketData: 040e04020c2000 +0ms
  hci   event type = 4 +0ms
  hci   sub event type = 14 +0ms
  hci           cmd = 8204 +0ms
  hci           status = 0 +0ms
  hci           result =  +0ms
  noble scanStart +10ms

But nothing more :(

sandeepmistry, 2015-08-17:
@pierreduf make sure reset command is processed before scanning is started, some things to try out:

1) Change: https://github.com/sandeepmistry/node-bluetooth-hci-socket/blob/master/lib/usb.js#L103 to check for: ```0e0402030c00```

2) Add a delay before scanning is started, using ```setTimeout``` with value of 1000ms or so

sandeepmistry, 2015-09-05:
@pierreduf ping, any updates?

sandeepmistry, 2015-09-20:
Closing for now due to lack of activity, feel free to open again with updates.

PR #31 (2016-06-27): Add support for Intel BLE Adapter
Should fix #3. Tested on 3 both Dell and Toshiba PCs with internal Intel BLE adapter, both running Windows 10.
commit ee8e7a19 (2016-06-27): Add support for Intel BLE Adapter 

Should fix #3. Tested on 3 both Dell and Toshiba PCs with internal Intel BLE adapter, both running Windows 10.
closed 2015-09-20 by sandeepmistry