noble/node-bluetooth-hci-socket #5: libudev-dev dependency

tompropst, 2015-08-27:
I was just attempting to install on a fresh Ubuntu system and discovered that it would fail until I installed libudev-dev.

sandeepmistry, 2015-08-29:
@tompropst thanks for reporting.

Was the ``node-usb`` dependency failing? It's optional, so shouldn't be an issue.

Could you please share the console output and npm-debug log of the failure?

tompropst, 2015-09-01:
@sandeepmistry, I don't recall which package the failure was occurring in.  I saw a specific reference to libudev.h not found, installed that dependency, and viola. I'll try to reproduce it.

tompropst, 2015-09-01:
@sandeepmistry here are the reproduction steps and error information.

1. Clean Ubuntu install.
1. Install Node.js.
   * ```sudo apt-get install nodejs```
1. Create `node` symlink to `nodejs`.
   * This was required to avoid an issue with gyp.
   * ```sudo ln -s /usr/bin/nodejs /usr/bin/node```
1. Install NPM.
   * ```sudo apt-get install npm```
1. Attempt `bluetooth-hci-socket` install.
   * ```npm install bluetooth-hci-socket```
```
npm http GET https://registry.npmjs.org/bluetooth-hci-socket
npm http 304 https://registry.npmjs.org/bluetooth-hci-socket
npm http GET https://registry.npmjs.org/debug
npm http GET https://registry.npmjs.org/usb
npm http GET https://registry.npmjs.org/nan
npm http 304 https://registry.npmjs.org/nan
npm http 304 https://registry.npmjs.org/debug
npm http 304 https://registry.npmjs.org/usb
npm http GET https://registry.npmjs.org/ms/0.7.1
npm http 304 https://registry.npmjs.org/ms/0.7.1

> usb@1.0.6 install /home/plarray/code/node_modules/bluetooth-hci-socket/node_modules/usb
> node-pre-gyp install --fallback-to-build

node-pre-gyp http GET https://tessel-builds.s3-us-west-2.amazonaws.com/pre-gyp/usb/v1.0.6/usb_bindings-v1.0.6-node-v11-linux-x64.tar.gz
node-pre-gyp http 404 https://tessel-builds.s3-us-west-2.amazonaws.com/pre-gyp/usb/v1.0.6/usb_bindings-v1.0.6-node-v11-linux-x64.tar.gz
node-pre-gyp http Pre-built binary not available for your system, looked for https://tessel-builds.s3-us-west-2.amazonaws.com/pre-gyp/usb/v1.0.6/usb_bindings-v1.0.6-node-v11-linux-x64.tar.gz (falling back to source compile with node-gyp) 
make: Entering directory `/home/plarray/code/node_modules/bluetooth-hci-socket/node_modules/usb/build'
  CC(target) Release/obj.target/libusb/libusb/libusb/core.o
  CC(target) Release/obj.target/libusb/libusb/libusb/descriptor.o
  CC(target) Release/obj.target/libusb/libusb/libusb/hotplug.o
  CC(target) Release/obj.target/libusb/libusb/libusb/io.o
  CC(target) Release/obj.target/libusb/libusb/libusb/strerror.o
  CC(target) Release/obj.target/libusb/libusb/libusb/sync.o
  CC(target) Release/obj.target/libusb/libusb/libusb/os/poll_posix.o
  CC(target) Release/obj.target/libusb/libusb/libusb/os/threads_posix.o
../libusb/libusb/os/threads_posix.c:24:0: warning: "_GNU_SOURCE" redefined [enabled by default]
 #  define _GNU_SOURCE
 ^
<command-line>:0:0: note: this is the location of the previous definition
  CC(target) Release/obj.target/libusb/libusb/libusb/os/linux_usbfs.o
  CC(target) Release/obj.target/libusb/libusb/libusb/os/linux_udev.o
../libusb/libusb/os/linux_udev.c:40:21: fatal error: libudev.h: No such file or directory
 #include <libudev.h>
                     ^
compilation terminated.
make: *** [Release/obj.target/libusb/libusb/libusb/os/linux_udev.o] Error 1
make: Leaving directory `/home/plarray/code/node_modules/bluetooth-hci-socket/node_modules/usb/build'
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/share/node-gyp/lib/build.js:267:23)
gyp ERR! stack     at ChildProcess.EventEmitter.emit (events.js:98:17)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (child_process.js:797:12)
gyp ERR! System Linux 3.19.0-26-generic
gyp ERR! command "nodejs" "/usr/bin/node-gyp" "build" "--fallback-to-build" "--module=/home/plarray/code/node_modules/bluetooth-hci-socket/node_modules/usb/src/binding/usb_bindings.node" "--module_name=usb_bindings" "--module_path=/home/plarray/code/node_modules/bluetooth-hci-socket/node_modules/usb/src/binding"
gyp ERR! cwd /home/plarray/code/node_modules/bluetooth-hci-socket/node_modules/usb
gyp ERR! node -v v0.10.25
gyp ERR! node-gyp -v v0.10.10
gyp ERR! not ok 
node-pre-gyp ERR! build error 
node-pre-gyp ERR! stack Error: Failed to execute 'node-gyp build --fallback-to-build --module=/home/plarray/code/node_modules/bluetooth-hci-socket/node_modules/usb/src/binding/usb_bindings.node --module_name=usb_bindings --module_path=/home/plarray/code/node_modules/bluetooth-hci-socket/node_modules/usb/src/binding' (1)
node-pre-gyp ERR! stack     at ChildProcess.<anonymous> (/home/plarray/code/node_modules/bluetooth-hci-socket/node_modules/usb/node_modules/node-pre-gyp/lib/util/compile.js:83:29)
node-pre-gyp ERR! stack     at ChildProcess.EventEmitter.emit (events.js:98:17)
node-pre-gyp ERR! stack     at maybeClose (child_process.js:743:16)
node-pre-gyp ERR! stack     at Process.ChildProcess._handle.onexit (child_process.js:810:5)
node-pre-gyp ERR! System Linux 3.19.0-26-generic
node-pre-gyp ERR! command "node" "/home/plarray/code/node_modules/bluetooth-hci-socket/node_modules/usb/node_modules/.bin/node-pre-gyp" "install" "--fallback-to-build"
node-pre-gyp ERR! cwd /home/plarray/code/node_modules/bluetooth-hci-socket/node_modules/usb
node-pre-gyp ERR! node -v v0.10.25
node-pre-gyp ERR! node-pre-gyp -v v0.6.4
node-pre-gyp ERR! not ok 
Failed to execute 'node-gyp build --fallback-to-build --module=/home/plarray/code/node_modules/bluetooth-hci-socket/node_modules/usb/src/binding/usb_bindings.node --module_name=usb_bindings --module_path=/home/plarray/code/node_modules/bluetooth-hci-socket/node_modules/usb/src/binding' (1)
npm WARN This failure might be due to the use of legacy binary "node"
npm WARN For further explanations, please read
/usr/share/doc/nodejs/README.Debian
 
npm WARN optional dep failed, continuing usb@1.0.6

> bluetooth-hci-socket@0.3.3 install /home/plarray/code/node_modules/bluetooth-hci-socket
> node-gyp rebuild

[... cut ...] th-hci-socket/build'
  CXX(target) Release/obj.target/binding/src/BluetoothHciSocket.o
  SOLINK_MODULE(target) Release/obj.target/binding.node
  SOLINK_MODULE(target) Release/obj.target/binding.node: Finished
  COPY Release/binding.node
make: Leaving directory `/home/plarray/code/node_modules/bluetooth-hci-socket/build'
bluetooth-hci-socket@0.3.3 node_modules/bluetooth-hci-socket
├── nan@2.0.8
└── debug@2.2.0 (ms@0.7.1)
```
At this point, installing `libudev-dev` and retrying results in success.

sandeepmistry, 2015-09-05:
@tompropst this is fine, as ```node-usb``` is an optional dependency on Linux, it's only needed on windows.

```sh
npm WARN optional dep failed, continuing usb@1.0.6
```

I've added a note in the read me for this.

closed 2015-09-05 by sandeepmistry