noble/node-bluetooth-hci-socket #6: fails to build on iojs 3.2.0  (on Fedora 22)

jrobeson, 2015-08-31:
I know the USB support is optional, but it probably shouldn't fail like this. 

```bash
$ iojs --version
v3.2.0
```

```bash
$ npm install bluetooth-hci-socket
-
> usb@1.0.6 install /home/johnny/.local/share/node_modules/bluetooth-hci-socket/node_modules/usb
> node-pre-gyp install --fallback-to-build

make: Entering directory '/home/johnny/.local/share/node_modules/bluetooth-hci-socket/node_modules/usb/build'
  CC(target) Release/obj.target/libusb/libusb/libusb/core.o
  CC(target) Release/obj.target/libusb/libusb/libusb/descriptor.o
  CC(target) Release/obj.target/libusb/libusb/libusb/hotplug.o
  CC(target) Release/obj.target/libusb/libusb/libusb/io.o
  CC(target) Release/obj.target/libusb/libusb/libusb/strerror.o
  CC(target) Release/obj.target/libusb/libusb/libusb/sync.o
  CC(target) Release/obj.target/libusb/libusb/libusb/os/poll_posix.o
  CC(target) Release/obj.target/libusb/libusb/libusb/os/threads_posix.o
../libusb/libusb/os/threads_posix.c:24:0: warning: "_GNU_SOURCE" redefined
 #  define _GNU_SOURCE
 ^
<command-line>:0:0: note: this is the location of the previous definition
  CC(target) Release/obj.target/libusb/libusb/libusb/os/linux_usbfs.o
  CC(target) Release/obj.target/libusb/libusb/libusb/os/linux_udev.o
  AR(target) Release/obj.target/usb.a
  COPY Release/usb.a
  CXX(target) Release/obj.target/usb_bindings/src/node_usb.o
In file included from ../src/helpers.h:3:0,
                 from ../src/node_usb.h:21,
                 from ../src/node_usb.cc:1:
../node_modules/nan/nan.h:324:27: error: redefinition of ‘template<class T> v8::Local<T> Nan::imp::NanEnsureHandleOrPersistent(const v8::Local<T>&)’
   NAN_INLINE v8::Local<T> NanEnsureHandleOrPersistent(const v8::Local<T> &val) {
                           ^
../node_modules/nan/nan.h:319:17: note: ‘template<class T> v8::Handle<T> Nan::imp::NanEnsureHandleOrPersistent(v8::Handle<T>&)’ previously declared here
   v8::Handle<T> NanEnsureHandleOrPersistent(const v8::Handle<T> &val) {
                 ^
../node_modules/nan/nan.h:344:27: error: redefinition of ‘template<class T> v8::Local<T> Nan::imp::NanEnsureLocal(v8::Handle<T>&)’
   NAN_INLINE v8::Local<T> NanEnsureLocal(const v8::Handle<T> &val) {
                           ^
../node_modules/nan/nan.h:334:27: note: ‘template<class T> v8::Local<T> Nan::imp::NanEnsureLocal(const v8::Local<T>&)’ previously declared here
   NAN_INLINE v8::Local<T> NanEnsureLocal(const v8::Local<T> &val) {
                           ^
../node_modules/nan/nan.h:757:13: error: ‘node::smalloc’ has not been declared
     , node::smalloc::FreeCallback callback
             ^
../node_modules/nan/nan.h:757:35: error: expected ‘,’ or ‘...’ before ‘callback’
     , node::smalloc::FreeCallback callback
                                   ^
../node_modules/nan/nan.h: In function ‘v8::Local<v8::Object> NanNewBufferHandle(char*, size_t, int)’:
../node_modules/nan/nan.h:761:50: error: ‘callback’ was not declared in this scope
         v8::Isolate::GetCurrent(), data, length, callback, hint);
                                                  ^
../node_modules/nan/nan.h:761:60: error: ‘hint’ was not declared in this scope
         v8::Isolate::GetCurrent(), data, length, callback, hint);
                                                            ^
../node_modules/nan/nan.h: In function ‘v8::Local<v8::Object> NanNewBufferHandle(const char*, uint32_t)’:
../node_modules/nan/nan.h:768:67: error: no matching function for call to ‘New(v8::Isolate*, const char*&, uint32_t&)’
     return node::Buffer::New(v8::Isolate::GetCurrent(), data, size);
                                                                   ^
In file included from ../src/node_usb.h:15:0,
                 from ../src/node_usb.cc:1:
/home/johnny/.node-gyp/3.2.0/include/node/node_buffer.h:31:40: note: candidate: v8::MaybeLocal<v8::Object> node::Buffer::New(v8::Isolate*, v8::Handle<v8::String>, node::encoding) <near match>
 NODE_EXTERN v8::MaybeLocal<v8::Object> New(v8::Isolate* isolate,
                                        ^
/home/johnny/.node-gyp/3.2.0/include/node/node_buffer.h:31:40: note:   conversion of argument 3 would be ill-formed:
In file included from ../src/helpers.h:3:0,
[... 43 lines cut ...]
node-pre-gyp ERR! stack     at emitTwo (events.js:87:13)
node-pre-gyp ERR! stack     at ChildProcess.emit (events.js:172:7)
node-pre-gyp ERR! stack     at maybeClose (internal/child_process.js:764:16)
node-pre-gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:211:5)
node-pre-gyp ERR! System Linux 4.1.6-200.fc22.x86_64
node-pre-gyp ERR! command "/usr/bin/iojs" "/home/johnny/.local/share/node_modules/bluetooth-hci-socket/node_modules/usb/node_modules/.bin/node-pre-gyp" "install" "--fallback-to-build"
node-pre-gyp ERR! cwd /home/johnny/.local/share/node_modules/bluetooth-hci-socket/node_modules/usb
node-pre-gyp ERR! node -v v3.2.0
node-pre-gyp ERR! node-pre-gyp -v v0.6.4
node-pre-gyp ERR! not ok 
Failed to execute '/usr/bin/iojs /usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js build --fallback-to-build --module=/home/johnny/.local/share/node_modules/bluetooth-hci-socket/node_modules/usb/src/binding/usb_bindings.node --module_name=usb_bindings --module_path=/home/johnny/.local/share/node_modules/bluetooth-hci-socket/node_modules/usb/src/binding' (1)
npm WARN optional dep failed, continuing usb@1.0.6

> bluetooth-hci-socket@0.3.3 install /home/johnny/.local/share/node_modules/bluetooth-hci-socket
> node-gyp rebuild

make: Entering directory '/home/johnny/.local/share/node_modules/bluetooth-hci-socket/build'
  CXX(target) Release/obj.target/binding/src/BluetoothHciSocket.o
  SOLINK_MODULE(target) Release/obj.target/binding.node
  COPY Release/binding.node
make: Leaving directory '/home/johnny/.local/share/node_modules/bluetooth-hci-socket/build'
bluetooth-hci-socket@0.3.3 node_modules/bluetooth-hci-socket
├── debug@2.2.0 (ms@0.7.1)
└── nan@2.0.8
```

sandeepmistry, 2015-09-05:
@jrobeson looks ok, the optional ```node-usb``` dependency is failing, I'll update the read me to mention this.

closed 2015-09-05 by sandeepmistry

paullryan, 2015-09-14:
@sandeepmistry This is also failing on windows which makes it unusable there as far as I can see. I know this isn't your issue as it's with the underlying node-usb library (from issue https://github.com/nonolith/node-usb/issues/89 and pull request https://github.com/nonolith/node-usb/pull/90) however it's something to note. Also why use raw usb instead of the rfcomm stack? What's missing from that stack?

sandeepmistry, 2015-09-20:
@paullryan RFCOMM is not an option for BLE, you need lower level L2CAP or HCI sockets. This module  only needs USB access on Windows, on Linux it uses kernel HCI sockets (hence the optional).

If there's a better way to specify a Windows only dependency in the package.json, please let me know.

paullryan, 2015-09-22:
@sandeepmistry Thanks for the answer on the needs for windows. I understand the dependency check and that's an acceptable mechanism at this point, I was just pointing out that with node v4.0.0 + it's currently failing on Windows given the nan2 needs for that. Again not your issue (it's a problem in node-usb) but just wanted to make you aware if it comes up again.

sandeepmistry, 2015-09-28:
@paullryan thanks for the clarification.

Does it make sense to create another issue to track Node.js 4.0+ not being able to work until node-usb issues are sorted out? Linux and OS X should be fine for 4.0+ as far as I can tell.

paullryan, 2015-09-28:
@sandeepmistry probably a good idea, added https://github.com/sandeepmistry/node-bluetooth-hci-socket/issues/7. 

And yes you're correct it looks like OS X and Linux are compiling and running fine with NaN2 and Node.js 4.x+

sandeepmistry, 2015-09-29:
@paullryan great, thanks for opening #7!