noble/node-bluetooth-hci-socket #9: Add support for BCM20702

PimButton, 2015-10-15:
Is it possible to add support for the BCM20702 chip as found in [this dongle](http://www.amazon.de/gp/product/B007MKMJGO/?ie=UTF8&camp=1638&creative=6742&linkCode=ur2&psc=1&redirect=true&site-redirect=de&tag=brandtlightin-21)  and the [ASUS BT-400](https://www.asus.com/us/Networking/USBBT400/) ? It works with Noble on Linux and the Mac but on Windows it won't. Since it can support up to 14 simultaneous connections it would be very nice to have this adapter supported on Windows as wel..

Kind regards!

dapetcu21, 2015-10-15:
+1 We have [this dongle](http://www.bestbuy.ca/en-CA/product/insignia-insignia-usb-bluetooth-adaptor-ns-pcy5bma-c-ns-pcy5bma-c/10292194.aspx?path=76fda8ee2797ba421eb12ec377d22c01en02) that uses BCM20702A0.

sandeepmistry, 2015-10-27:
@PimButton @dapetcu21 no issue adding it, can you please provide the USB VID and PID? (``lsusb`` on Linux can provide this info.)

PimButton, 2015-10-27:
@sandeepmistry great, will provide the info ASAP.

PimButton, 2015-10-27:
Ok, here is the info for the Asus BT-400, I don't have another variant of this dongle so can't check if the VID is different for another manufacturer. 

USB PID: 0x17cb
USB VID: 0x0b05 (ASUSTek Computer Inc.)

sandeepmistry, 2015-10-27:
@PimButton thanks I've added in https://github.com/sandeepmistry/node-bluetooth-hci-socket/commit/999fbdd4f19fa8069e505686e56b46e2ccf87058

Please open a new issue if you ever find the PID/VID for the other adapter cc/ @dapetcu21

closed 2015-10-27 by sandeepmistry