noble/node-bluetooth-hci-socket #12: Support for mac

mhle, 2015-11-16:
Are you planning to support Mac as well in the future? OS X el capital currently doesn't support more than 6 connected devices maybe using this lib can resolve that issue?

sandeepmistry, 2015-11-28:
@mhle no plans to add support.

However, it is possible to use it if you build from source. Then you have to unload the built-in OS X driver for the adapter, so that's a big drag.

closed 2015-11-28 by sandeepmistry

alexsuse, 2017-04-22:
Any chance you can provide a bit more context on how to "build from source"? I assume you just need to compile the cpp files, but given that the package seems to bind those calls in js, it might be a bit more complicated.

sandeepmistry, 2017-04-23:
@alexsuse you just need `node-usb` actually - the native bindings are not used at all.