noble/node-bluetooth-hci-socket #13: Bluetooth dongle not detected

mhle, 2015-11-20:
I'm running Noble in an Electron app but sometimes the dongle we're using is not detected and so onStateChanged is not triggered. 

After digging into the code we found that BluetoothHciSocket.prototype.isDevUp returns false even if the device is plugged in. 
https://github.com/sandeepmistry/node-bluetooth-hci-socket/blob/master/lib/usb.js#L68

We're using a BCM20702A0 dongle with PID 21E8 and VID 0A5C which should be supported.

Is there any way to recover from this? or force detection of the dongle again?

sandeepmistry, 2015-11-28:
@mhle please provide debug logs, by setting the ```DEBUG``` env var to ```*```.

I'm thinking the adapter is not responding to the HCI reset command, but we'll see what the debug logs say.

boredSinan, 2015-12-03:
I have the same problem with my BLE USB Dongle (chipset: CSR8510 A10).

My code always communicates with the dongle only every 2nd time!!
I found out that although 'this._isUp' is true, it still does not work. The function noble.on() won't run.

In the following you can see the debug output for the case when there is no communication with the dongle:

[image]

Thank you for your helps..

boredSinan, 2015-12-03:
And here is the debug output when there is a communication to the dongle:

[image]

boredSinan, 2015-12-03:
I am still trying to solve this problem. It is strange, that SanDeep's "enter-exit.js" code always runs.
My code is based on "enter-exit.js", so I do not understand why I got this problem? Because I did not change the code in its basic functionality.

Here is the debug output of "enter-exit.js" to compare with those above.
[image]

sandeepmistry, 2015-12-04:
@boredSinan looks like the adapter is still scanning when you start you app the 2nd time, and then doesn't respond to any HCI commands after the reset command.

How are you exiting your app? noble tries to stop scanning on exit here: https://github.com/sandeepmistry/noble/blob/master/lib/hci-socket/bindings.js#L96

Another thing to try is, adding a delay here, to give the adapter some time before we send other HCI commands:  https://github.com/sandeepmistry/noble/blob/master/lib/hci-socket/bindings.js#L96

```javascript
setTimeout(function() {
  this._isUp = true;
}.bind(this), 500);
```

PS: I like the coloured screenshots, but please also copy/paste the test next time, it's easier to search etc.

boredSinan, 2015-12-07:
Thanks a lot Sandeep, for your suggestions.
I found out my problem: There is the following function in my code:

//Wait function in ms
function pauseMs(millis) {
    var date = new Date();
    var curDate = null;
    do {
        curDate = new Date();
    }
    while (curDate - date < millis);
} 

And the problem always occured when I closed the Node.js application at the same time as the code was running the pauseMs() function (this function was called several times in my code). Because of that, the onExit() function could not be executed.

Now I don't close the Node.js application at the same time as the application is paused (because of pauseMs()), there are no problems anymore!

boredSinan, 2015-12-07:
Thanks a lot Sandeep, for your suggestions.
I found out my problem: There is the following function in my code:

//Wait function in ms
function pauseMs(millis) {
var date = new Date();
var curDate = null;
do {
curDate = new Date();
}
while (curDate - date < millis);
}

And the problem always occured when I closed the Node.js application at the
same time as the code was running the pauseMs() function (this function was
called several times in my code). Because of that, the onExit() function
could not be executed.

Now I don't close the Node.js application at the same time as the
application is paused (because of pauseMs()), there are no problems anymore!

I uploaded this post on:
https://github.com/sandeepmistry/node-bluetooth-hci-socket/issues/13

sandeepmistry, 2015-12-08:
@boredSinan cool, thanks for following up!

@mhle any updates from my request in https://github.com/sandeepmistry/node-bluetooth-hci-socket/issues/13#issuecomment-160309753

sandeepmistry, 2016-01-03:
Closing due to lack of activity.

closed 2016-01-03 by sandeepmistry