noble/node-bluetooth-hci-socket #15: Crypto error when connecting to WICED Sense

deadprogram, 2015-12-19:
When connecting to the Broadcom WICED Sense tag from Linux (Ubuntu 14.04), I hit the following error. I do not have this problem with any other BLE hardware:

```
buffer.js:229
      length += list[i].length;
                       ^

TypeError: Cannot read property 'length' of undefined
    at Function.Buffer.concat (buffer.js:229:24)
    at Object.c1 (/home/ron/Development/cylonjs/cylon-example/node_modules/cylon-ble/node_modules/noble/lib/hci-socket/crypto.js:8:19)
    at Smp.handlePairingResponse (/home/ron/Development/cylonjs/cylon-example/node_modules/cylon-ble/node_modules/noble/lib/hci-socket/smp.js:86:12)
    at Smp.onAclStreamData (/home/ron/Development/cylonjs/cylon-example/node_modules/cylon-ble/node_modules/noble/lib/hci-socket/smp.js:57:10)
    at emitTwo (events.js:92:20)
    at emit (events.js:172:7)
    at AclStream.push (/home/ron/Development/cylonjs/cylon-example/node_modules/cylon-ble/node_modules/noble/lib/hci-socket/acl-stream.js:35:10)
    at NobleBindings.onAclDataPkt (/home/ron/Development/cylonjs/cylon-example/node_modules/cylon-ble/node_modules/noble/lib/hci-socket/bindings.js:252:15)
    at emitThree (events.js:97:13)
    at emit (events.js:175:7)
```

sandeepmistry, 2016-01-03:
@deadprogram this seems like an issue in noble, could you please "move" the issue there (https://github.com/sandeepmistry/noble/issues).

As well as include what you've done so far to debug the issue ...

deadprogram, 2016-01-03:
Hi, @sandeepmistry will do, right after I confirm also exists on OSX.

deadprogram, 2016-01-03:
Replace by https://github.com/sandeepmistry/noble/issues/295 closing.

closed 2016-01-03 by deadprogram