noble/node-bluetooth-hci-socket #17: install bluetooth-hci-socket on mips.

kaiser-ren, 2016-01-18:
Hello,
I have a wrtnode2r which should be openwrt platform and mips ARCH, I want to install bluetooth-hci-socket on it, but below is error log, can you help me? Thanks.
root@OpenWrt:~/node_modules/noble# npm install bluetooth-hci-socket
npm WARN engine usb@1.1.1: wanted: {"node":">=0.12.x"} (current: {"node":"0.10.36","npm":"1.4.28"})

> usb@1.1.1 install /root/node_modules/noble/node_modules/bluetooth-hci-socket/node_modules/usb
> node-pre-gyp install --fallback-to-build

gyp ERR! configure error
gyp ERR! stack Error: Can't find Python executable "python", you can set the PYTHON env variable.
gyp ERR! stack     at failNoPython (/usr/lib/node_modules/npm/node_modules/node-gyp/lib/configure.js:103:14)
gyp ERR! stack     at /usr/lib/node_modules/npm/node_modules/node-gyp/lib/configure.js:42:11
gyp ERR! stack     at F (/usr/lib/node_modules/npm/node_modules/which/which.js:43:25)
gyp ERR! stack     at E (/usr/lib/node_modules/npm/node_modules/which/which.js:46:29)
gyp ERR! stack     at /usr/lib/node_modules/npm/node_modules/which/which.js:57:16
gyp ERR! stack     at Object.oncomplete (evalmachine.<anonymous>:108:15)
gyp ERR! System Linux 3.18.23
gyp ERR! command "/usr/bin/nodejs" "/usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "configure" "--fallback-to-build" "--module=/root/node_modules/noble/node_modules/bluetooth-hci-socket/node_modules/usb/src/binding/usb_bindings.node" "--module_name=usb_bindings" "--module_path=/root/node_modules/noble/node_modules/bluetooth-hci-socket/node_modules/usb/src/binding"
gyp ERR! cwd /root/node_modules/noble/node_modules/bluetooth-hci-socket/node_modules/usb
gyp ERR! node -v v0.10.36
gyp ERR! node-gyp -v v1.0.1
gyp ERR! not ok
node-pre-gyp ERR! build error
node-pre-gyp ERR! stack Error: Failed to execute '/usr/bin/nodejs /usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js configure --fallback-to-build --module=/root/node_modules/noble/node_modules/bluetooth-hci-socket/node_modules/usb/src/binding/usb_bindings.node --module_name=usb_bindings --module_path=/root/node_modules/noble/node_modules/bluetooth-hci-socket/node_modules/usb/src/binding' (1)
node-pre-gyp ERR! stack     at ChildProcess.<anonymous> (/root/node_modules/noble/node_modules/bluetooth-hci-socket/node_modules/usb/node_modules/node-pre-gyp/lib/util/compile.js:83:29)
node-pre-gyp ERR! stack     at ChildProcess.emit (events.js:98:17)
node-pre-gyp ERR! stack     at maybeClose (child_process.js:766:16)
node-pre-gyp ERR! stack     at Process.ChildProcess._handle.onexit (child_process.js:833:5)
node-pre-gyp ERR! System Linux 3.18.23
node-pre-gyp ERR! command "node" "/root/node_modules/noble/node_modules/bluetooth-hci-socket/node_modules/usb/node_modules/.bin/node-pre-gyp" "install" "--fallback-to-build"
node-pre-gyp ERR! cwd /root/node_modules/noble/node_modules/bluetooth-hci-socket/node_modules/usb
node-pre-gyp ERR! node -v v0.10.36
node-pre-gyp ERR! node-pre-gyp -v v0.6.4
node-pre-gyp ERR! not ok
Failed to execute '/usr/bin/nodejs /usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js configure --fallback-to-build --module=/root/node_modules/noble/node_modules/bluetooth-hci-socket/node_modules/usb/src/binding/usb_bindings.node --module_name=usb_bindings --module_path=/root/node_modules/noble/node_modules/bluetooth-hci-socket/node_modules/usb/src/binding' (1)
npm WARN optional dep failed, continuing usb@1.1.1

> bluetooth-hci-socket@0.4.2 install /root/node_modules/noble/node_modules/bluetooth-hci-socket
> node-gyp rebuild

gyp ERR! configure error
gyp ERR! stack Error: Can't find Python executable "python", you can set the PYTHON env variable.
gyp ERR! stack     at failNoPython (/usr/lib/node_modules/npm/node_modules/node-gyp/lib/configure.js:103:14)
gyp ERR! stack     at /usr/lib/node_modules/npm/node_modules/node-gyp/lib/configure.js:42:11
gyp ERR! stack     at F (/usr/lib/node_modules/npm/node_modules/which/which.js:43:25)
gyp ERR! stack     at E (/usr/lib/node_modules/npm/node_modules/which/which.js:46:29)
gyp ERR! stack     at /usr/lib/node_modules/npm/node_modules/which/which.js:57:16
gyp ERR! stack     at Object.oncomplete (evalmachine.<anonymous>:108:15)
gyp ERR! System Linux 3.18.23
gyp ERR! command "node" "/usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /root/node_modules/noble/node_modules/bluetooth-hci-socket
gyp ERR! node -v v0.10.36
gyp ERR! node-gyp -v v1.0.1
gyp ERR! not ok
npm WARN optional dep failed, continuing bluetooth-hci-socket@0.4.2

jrobeson, 2016-01-19:
you need python for node-pre-gyp. the module won't work without it.

kaiser-ren, 2016-01-19:
Hi Jrobeson,
I have installed python, this is what i got after "opkg list-installed":
**python - 2.7.9-5
python-base - 2.7.9-5
python-codecs - 2.7.9-5
python-compiler - 2.7.9-5
python-ctypes - 2.7.9-5
python-db - 2.7.9-5
python-decimal - 2.7.9-5
python-distutils - 2.7.9-5
python-email - 2.7.9-5
python-gdbm - 2.7.9-5
python-light - 2.7.9-5
python-logging - 2.7.9-5
python-multiprocessing - 2.7.9-5
python-ncurses - 2.7.9-5
python-openssl - 2.7.9-5
python-pydoc - 2.7.9-5
python-sqlite3 - 2.7.9-5
python-unittest - 2.7.9-5
python-xml - 2.7.9-5**

But the result is the same:
**npm WARN engine usb@1.1.1: wanted: {"node":">=0.12.x"} (current: {"node":"0.10.36","npm":"1.4.28"})
-
> usb@1.1.1 install /root/node_modules/bluetooth-hci-socket/node_modules/usb
> node-pre-gyp install --fallback-to-build

gyp WARN install got an error, rolling back install
gyp ERR! UNCAUGHT EXCEPTION
gyp ERR! stack Error: ENOENT, lstat '/root/.node-gyp/0.10.36/deps/openssl/openssl/ms/uplink.h'
gyp ERR! System Linux 3.18.23
gyp ERR! command "/usr/bin/nodejs" "/usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "configure" "--fallback-to-build" "--module=/root/node_modules/bluetooth-hci-socket/node_modules/usb/src/binding/usb_bindings.node" "--module_name=usb_bindings" "--module_path=/root/node_modules/bluetooth-hci-socket/node_modules/usb/src/binding"
gyp ERR! cwd /root/node_modules/bluetooth-hci-socket/node_modules/usb
gyp ERR! node -v v0.10.36
gyp ERR! node-gyp -v v1.0.1
gyp ERR! This is a bug in `node-gyp`.
gyp ERR! Try to update node-gyp and file an Issue if it does not help:
gyp ERR!     <https://github.com/TooTallNate/node-gyp/issues>
node-pre-gyp ERR! build error
node-pre-gyp ERR! stack Error: Failed to execute '/usr/bin/nodejs /usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js configure --fallback-to-build --module=/root/node_modules/bluetooth-hci-socket/node_modules/usb/src/binding/usb_bindings.node --module_name=usb_bindings --module_path=/root/node_modules/bluetooth-hci-socket/node_modules/usb/src/binding' (7)
node-pre-gyp ERR! stack     at ChildProcess.<anonymous> (/root/node_modules/bluetooth-hci-socket/node_modules/usb/node_modules/node-pre-gyp/lib/util/compile.js:83:29)
node-pre-gyp ERR! stack     at ChildProcess.emit (events.js:98:17)
node-pre-gyp ERR! stack     at maybeClose (child_process.js:766:16)
node-pre-gyp ERR! stack     at Process.ChildProcess._handle.onexit (child_process.js:833:5)
node-pre-gyp ERR! System Linux 3.18.23
node-pre-gyp ERR! command "node" "/root/node_modules/bluetooth-hci-socket/node_modules/usb/node_modules/.bin/node-pre-gyp" "install" "--fallback-to-build"
node-pre-gyp ERR! cwd /root/node_modules/bluetooth-hci-socket/node_modules/usb
node-pre-gyp ERR! node -v v0.10.36
node-pre-gyp ERR! node-pre-gyp -v v0.6.4
node-pre-gyp ERR! not ok
Failed to execute '/usr/bin/nodejs /usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js configure --fallback-to-build --module=/root/node_modules/bluetooth-hci-socket/node_modules/usb/src/binding/usb_bindings.node --module_name=usb_bindings --module_path=/root/node_modules/bluetooth-hci-socket/node_modules/usb/src/binding' (7)
npm WARN optional dep failed, continuing usb@1.1.1

> bluetooth-hci-socket@0.4.2 install /root/node_modules/bluetooth-hci-socket
> node-gyp rebuild

gyp WARN install got an error, rolling back install
gyp ERR! configure error
gyp ERR! stack Error: read ECONNRESET
gyp ERR! stack     at errnoException (net.js:905:11)
gyp ERR! stack     at TCP.onread (net.js:559:19)
gyp ERR! System Linux 3.18.23
gyp ERR! command "node" "/usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /root/node_modules/bluetooth-hci-socket
gyp ERR! node -v v0.10.36
gyp ERR! node-gyp -v v1.0.1
gyp ERR! not ok

npm ERR! bluetooth-hci-socket@0.4.2 install: `node-gyp rebuild`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the bluetooth-hci-socket@0.4.2 install script.
npm ERR! This is most likely a problem with the bluetooth-hci-socket package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-gyp rebuild
npm ERR! You can get their info via:
npm ERR!     npm owner ls bluetooth-hci-socket
npm ERR! There is likely additional logging output above.
npm ERR! System Linux 3.18.23
npm ERR! command "node" "/usr/bin/npm" "install" "bluetooth-hci-socket"
npm ERR! cwd /root
npm ERR! node -v v0.10.36
npm ERR! npm -v 1.4.28
npm ERR! code ELIFECYCLE
npm ERR! not ok code 0**

jrobeson, 2016-01-19:
it seems pretty clear why it isn't getting farther. do you see the node version error?

sandeepmistry, 2016-01-23:
@jrobeson thanks for helping out.

@kaiser-ren please try to read the logs next time, as the result is not "the same". For some reason this files is missing:
```
gyp ERR! stack Error: ENOENT, lstat '/root/.node-gyp/0.10.36/deps/openssl/openssl/ms/uplink.h'
gyp ERR! System Linux 3.18.23
```

Maybe you are out of disk space?

Also, make your your system has the required ```node-gyp``` dependencies: https://github.com/nodejs/node-gyp#installation

sandeepmistry, 2016-01-30:
Closing due to lack of activity ...

closed 2016-01-30 by sandeepmistry