noble/node-bluetooth-hci-socket #18: Support for multiple usb adapters in raw USB mode (with same VID and PID)?

NPohl, 2016-03-09:
Hi,

I'm using multiple adapters simultaneously to communicate with more than 5 devices at the same time. For this I'm using the raw mode, since it is much more stable than with bluez. To differentiate the adapters, I'm using USB dongles with different VID/PIDs and BLUETOOTH_HCI_SOCKET_USB_VID and BLUETOOTH_HCI_SOCKET_USB_PID to according values. 

This works well, but it is limited to knowing the exact VID/PIDs. It would be much nicer to enumerate the adapters with a simple number (like in noble with the NOBLE_HCI_DEVICE_ID environment variable) which also would allow to use multiple adapters with same VID and PID. Also it would be nice to enumerate valid values for this number, so a program can easily discover all available BLE adapters and use them.

Thanks,
Norman

sandeepmistry, 2016-03-13:
@NPohl this is not an easy problem to solve, each noble instance is not persistent like the kernel. The interface number can change, for example if you remove or add an adapter.

I could look into merging the [noble bind-user](https://github.com/sandeepmistry/noble/tree/bind-user) branch, then you can use noble in "user" mode, and the kernel will not interfere.

NPohl, 2016-03-16:
Oh ok, that's a problem I haven't thought of. I see that keeping such an ID valid over time would be hard. But maybe that isn't necessary, if there is something which would allow to differentiate adapters of the same type:

I've digged a bit into the usb protocol and found that devices also can have a serial number. The serial number could be set as a third environment variable (e.g.  BLUETOOTH_HCI_SOCKET_USB_SN) to identify which exact device should be used by the current process. 

To enumerate the connected adapters, a function could return a list of tuples with the VID/PID/SN of the connected adapters.

Do you think that could work?

sandeepmistry, 2016-03-20:
> Do you think that could work?

Yes, sounds good. Looks like node-usb exposes the serial number: https://github.com/nonolith/node-usb#devicedescriptor

It would be great if you could dive in and submit a pull request to add support for this.

NPohl, 2016-04-08:
Sorry, that took me longer than I've thought. It turned out, that
a) not all USB devices necessarily have a serial number
b) the serial number needs to be queried, which turns out to be an async operation (which would complicate things also a lot).

Instead I went for a combination of the USB bus number and the USB address, since these two together also identify an USB device uniquely (at least until reboot/disconnected or something like that).

These numbers can be set using the following environment variables:
BLUETOOTH_HCI_SOCKET_USB_BUS 
BLUETOOTH_HCI_SOCKET_USB_ADDRESS

The numbers for these can be easily acquired with the lsusb command.

NPohl, 2016-04-08:
These environment variables need to be set additionally to BLUETOOTH_HCI_SOCKET_USB_VID and BLUETOOTH_HCI_SOCKET_USB_PID, so if one wants to use this, the following environment variables need to be set:

BLUETOOTH_HCI_SOCKET_FORCE_USB = 1
BLUETOOTH_HCI_SOCKET_USB_VID = <usb vid> 
BLUETOOTH_HCI_SOCKET_USB_PID = <usb pid>
BLUETOOTH_HCI_SOCKET_USB_BUS = <usb bus>
BLUETOOTH_HCI_SOCKET_USB_ADDRESS = <usb address>

All these values can be obtained with the lsusb command or with the following function

```
var supportedIds = [
    { vid: 0x0a5c, pid: 0x21e8 },
    { vid: 0x19ff, pid: 0x0239 },
    { vid: 0x0a12, pid: 0x0001 },
    { vid: 0x0b05, pid: 0x17cb }
];

function getSupportedDeviceIds() {
    var deviceIds = [];

    var usbDevices = usb.getDeviceList()
    for (var i = 0; i < usbDevices.length; i++) {
      var usbDeviceDesc = usbDevices[i].deviceDescriptor

      for (var idxSupportedIds = 0; idxSupportedIds < supportedIds.length; ++idxSupportedIds) {
        var usbVid = supportedIds[idxSupportedIds].vid;
        var usbPid = supportedIds[idxSupportedIds].pid;

        if ((usbDeviceDesc.idVendor == usbVid) &&
            (usbDeviceDesc.idProduct == usbPid)) {
          deviceIds.push({
            bus: usbDevices[i].busNumber,
            address: usbDevices[i].deviceAddress,
            vid: usbDeviceDesc.idVendor,
            pid: usbDeviceDesc.idProduct
          });
        }
      }
    }
    
    return deviceIds;
};

```

I'm not sure where to put this function, yet. I use it in my own code right now. But I think it would make sense to include it somehow - to keep the list of supported vid/pids updated.

jeberhardt, 2016-04-18:
Could you provide some details on how you use noble and node-Bluetooth-hci-socket together in raw mode?  I'm having problems on Raspberry Pi, and think this might help..

Thanks!

sandeepmistry, 2016-04-23:
Functionality for this added in #21 by @NPohl.

sandeepmistry, 2016-04-23:
@jeberhardt please see: https://github.com/sandeepmistry/node-bluetooth-hci-socket#linux-1

closed 2016-04-23 by sandeepmistry

sivann-tw, 2017-01-16:
Hi Sandeepmistry,
  We are trying to bring up 5 BLE USB adapters in separate nodejs processes as well. However, there is some issue and we are still figuring out. My question is that since which version does noble start support multiple USB adapter? 
Thanks.

sandeepmistry, 2017-01-30:
@sivann-tw not specific to noble per say, you just need to use `v0.4.2` or later of `bluetooth-hci-socket` with noble.