noble/node-bluetooth-hci-socket #20: discoverAllServicesAndCharacteristics stopped working with raw USB mode at version 0.4.3

NPohl, 2016-04-08:
I cannot discover services and characteristics while using the raw USB mode anymore. It worked (and still works) with version 0.4.2 though. These are the adapters I've tested this with (separately):

Bus 001 Device 006: ID 0a12:0001 Cambridge Silicon Radio, Ltd Bluetooth Dongle (HCI mode)
Bus 001 Device 008: ID 0a5c:21e8 Broadcom Corp. BCM20702A0 Bluetooth 4.0

sandeepmistry, 2016-04-11:
@NPohl could you please provide some debug logs?
 * ```sudo DEBUG=* node <file>.js```
 * ```sudo hcidump -t -x```

This is probably related to https://github.com/sandeepmistry/noble/issues/372 as well. I'm trying to track it down, but haven't reproduced the issue locally yet.

sandeepmistry, 2016-04-11:
Fixed via https://github.com/sandeepmistry/node-bluetooth-hci-socket/commit/25999ec523325e43d5d900a7ae8866607f56fd9b and which had been published to npm as v0.4.4.

closed 2016-04-11 by sandeepmistry