noble/node-bluetooth-hci-socket #24: bluetooth-hci-socket can't find my bluetooth device

Galch, 2016-04-18:
i wrote some program using ble, but it only not working in my new laptop.

i wonder that this library can't connect to BLE 4.1 device?

### enviromnet
- Dell XPS13 9350
 - ble driver : Dell Wireless 1820A Bluetooth 4.1 LE

### error log
\node_modules\bluetooth-hci-socket\lib\usb.js:50
    throw new Error('No compatible USB Bluetooth 4.0 device found!');
    ^

Error: No compatible USB Bluetooth 4.0 device found!
    at BluetoothHciSocket.bindUser (C:\Users\Nacer\Documents\1.nBand\BandLogger-master\BandLogger-master\node_modules\bluetooth-hci-socket\lib\usb.js:50:11)
    at BluetoothHciSocket.bindRaw (C:\Users\Nacer\Documents\1.nBand\BandLogger-master\BandLogger-master\node_modules\bluetooth-hci-socket\lib\usb.js:28:8)
    at Hci.init (C:\Users\Nacer\Documents\1.nBand\BandLogger-master\BandLogger-master\node_modules\noble\lib\hci-socket\hci.js:89:16)
    at NobleBindings.init (C:\Users\Nacer\Documents\1.nBand\BandLogger-master\BandLogger-master\node_modules\noble\lib\hci-socket\bindings.js:83:13)
    at new Noble (C:\Users\Nacer\Documents\1.nBand\BandLogger-master\BandLogger-master\node_modules\noble\lib\noble.js:67:18)
    at Object.<anonymous> (C:\Users\Nacer\Documents\1.nBand\BandLogger-master\BandLogger-master\node_modules\noble\index.js:3:18)
    at Module._compile (module.js:409:26)
    at Object.Module._extensions..js (module.js:416:10)
    at Module.load (module.js:343:32)
    at Function.Module._load (module.js:300:12)

sandeepmistry, 2016-04-24:
@Galch only the following adapters are whitelisted now:

https://github.com/sandeepmistry/node-bluetooth-hci-socket#compatible-bluetooth-40-usb-adapters

 You can try adding the USB id's of yours here: https://github.com/sandeepmistry/node-bluetooth-hci-socket/blob/master/lib/usb.js#L66, if it works please submit a pull request so it can be added to the whitelist.

sandeepmistry, 2016-05-09:
I'm closing this for now, due to lack of discussion.

closed 2016-05-09 by sandeepmistry

gfwilliams, 2016-11-03:
Same issue here - as far as I can tell the Atheros QCA6174 that does BLE (and Wifi) isn't a USB device. And even if I add a BLE USB device that I know works, Windows won't enable it because it knows there i already an active BLE adaptor!