noble/node-bluetooth-hci-socket #30: [Windows 8

PimButton, 2016-06-23:


PimButton, 2016-07-10:
Hi @sandeepmistry, I know you are very busy so I don't want to pressure you in any way. I you can tell me if setting DEBUG causes some sort of extra reset or does something else that I can trigger manually it would be of great help so I can continue testing. 
Thanks for the amazing work, again!

sandeepmistry, 2016-07-13:
@PimButton debug will have impact on the timing.

Could you please try the latest, node-bluetooth-hci-socket version. I'm wondering if the following change has any impact: https://github.com/sandeepmistry/node-bluetooth-hci-socket/pull/31/files

PimButton, 2016-07-18:
@sandeepmistry thanks for your reply!
Unfortunately even with version 0.5.0 the same behaviors occurs. It seems about 50/50 whether the ``state`` is poweredOff or poweredOn. I also tried calling ``noble.stopScanning()`` before called the onStateChange function but that has no effect as well...

Asoustek, 2016-07-23:
I am also seeing this issue, same scenario after multiple runs sometimes i never get the state changed event, but in DEBUG=* mode i do.. I'm running on Windows 10 with the latest version of noble..

After some digging i've noticed that adding this bit of code to the BluetoothHciSocket.write function gives me (seemingly) 100% consistency... Now admittedly i don't know _anything_ about how this code works so maybe this just resolves the timing issue on my machine only.

```
var lastWrite = 0;
BluetoothHciSocket.prototype.write = function(data) {
  var waitTime = 0;
  var thisWrite = new Date().getTime();
  if (lastWrite != 0)
  {
    if (thisWrite - lastWrite < 5)
    {
      waitTime = 5;
    }
  }
  lastWrite = thisWrite;

  var waitTimer = thisWrite;
  while(new Date().getTime() - waitTimer < waitTime){}

  if (this._mode === 'raw' || this._mode === 'user') {
    var type = data.readUInt8(0);

    if (HCI_COMMAND_PKT === type) {
      this._usbDevice.controlTransfer(usb.LIBUSB_REQUEST_TYPE_CLASS | usb.LIBUSB_RECIPIENT_INTERFACE, 0, 0, 0, data.slice(1));
    } else if(HCI_ACLDATA_PKT === type) {
      this._aclDataOutEndpoint.transfer(data.slice(1));
    }
  }
};
```

PimButton, 2016-07-25:
@Asoustek, sounds promising, thanks for your reply! I am not able to test this week but once I'm back at work I will give it a try. 
Curious if this might give @sandeepmistry a clue about what's going on.

sandeepmistry, 2016-07-27:
Looking at the [node-usb API docs](https://github.com/nonolith/node-usb#api) both `this._usbDevice.controlTransfer` and `this._aclDataOutEndpoint.transfer` accept callback parameters.

@PimButton @Asoustek it would be great if you could look into queuing both ACL and command transfer data. Basically, if the callback has not being called for each, do not transfer the data until it is.

Asoustek, 2016-07-28:
Submitted https://github.com/sandeepmistry/node-bluetooth-hci-socket/pull/33

Unfortunately unable to test it, maybe once i get back to my bluetooth machine. You could add consoleLog in the onWriteFinished inside the if to make sure we're clearing the buffer.

If @PimButton or @sandeepmistry have time to pull the PR and try it i'd appreciate, otherwise i'll test my own a bit later.

PimButton, 2016-08-05:
I'm back at the office and was able to test :)
@Asoustek thanks for the quick PR! I pulled it, but unfortunately it did not work for me, the same behavior still occurred. However, the piece of code that you wrote above **does** work for me! As far as I can see you implemented the changes suggested by @sandeepmistry correctly so maybe something else is going on? Sandeep, do you understand why the manual time delay does work, but the callback does not?

Asoustek, 2016-08-09:
@PimButton yeah unfortunately i was seeing similar things, obeying the callbacks still occasionally gives me a weird state, but the actual timeout does seem to work very consistently for me.. Not sure if my PR is wrong, but i think it looks ok.

Maybe @sandeepmistry knows.

sandeepmistry, 2016-08-22:
I would suggest trying to add a `process.nextTick(function() { ... }` or `setTimeout` in the `onWriteFinished` function.

sandeepmistry, 2016-09-26:
@PimButton @Asoustek ping ... any progress on this?

sandeepmistry, 2016-10-16:
Closing this due to lack of activity and progress for now.

closed 2016-10-16 by sandeepmistry