noble/node-bluetooth-hci-socket #35: Installing on Windows 10

mrothenbuecher, 2016-08-30:
Hi,

i have following issue when I try to install.
```
PS C:\nuimo-electron> npm install bluetooth-hci-socket
npm WARN package.json nuimo-electron@0.0.1 No repository field.
npm WARN package.json nuimo-electron@0.0.1 No README data
-
> usb@1.1.2 install C:\nuimo-electron\node_modules\bluetooth-hci-socket\node_modules\usb
> node-pre-gyp install --fallback-to-build

[usb] Success: "C:\nuimo-electron\node_modules\bluetooth-hci-socket\node_modules\usb\src\binding\usb_bindings.node" is installed via remote

> bluetooth-hci-socket@0.5.0 install C:\nuimo-electron\node_modules\bluetooth-hci-socket
> node-gyp rebuild

C:\nuimo-electron\node_modules\bluetooth-hci-socket>if not defined npm_config_node_gyp (node "C:\Program Files\nodejs\node_modules\npm\bin\node-gyp-bin\\..\..\node_modules\node-gyp\bin\node-gyp.js" rebuild )  else (node ""
Die Projekte in dieser Projektmappe werden nacheinander erstellt. Um eine parallele Erstellung zu ermöglichen, müssen Sie den Schalter "/m" hinzufügen.
  win_delay_load_hook.c
C:\Program Files\nodejs\node_modules\npm\node_modules\node-gyp\src\win_delay_load_hook.c(34): error C2373: "__pfnDliNotifyHook2": Neudefinition; unterschiedliche Modifizierer [C:\nuimo-electron\node_modules\bluetooth-hci-s
  C:\Program Files (x86)\Microsoft Visual Studio 14.0\VC\include\delayimp.h(134): note: Siehe Deklaration von "__pfnDliNotifyHook2"
gyp ERR! build error
gyp ERR! stack Error: `C:\Program Files (x86)\MSBuild\14.0\bin\msbuild.exe` failed with exit code: 1
gyp ERR! stack     at ChildProcess.onExit (C:\Program Files\nodejs\node_modules\npm\node_modules\node-gyp\lib\build.js:276:23)
gyp ERR! stack     at emitTwo (events.js:87:13)
gyp ERR! stack     at ChildProcess.emit (events.js:172:7)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:200:12)
gyp ERR! System Windows_NT 10.0.10586
gyp ERR! command "C:\\Program Files\\nodejs\\node.exe" "C:\\Program Files\\nodejs\\node_modules\\npm\\node_modules\\node-gyp\\bin\\node-gyp.js" "rebuild"
gyp ERR! cwd C:\nuimo-electron\node_modules\bluetooth-hci-socket
gyp ERR! node -v v4.4.4
gyp ERR! node-gyp -v v3.3.1
gyp ERR! not ok
npm ERR! Windows_NT 10.0.10586
npm ERR! argv "C:\\Program Files\\nodejs\\node.exe" "C:\\Program Files\\nodejs\\node_modules\\npm\\bin\\npm-cli.js" "install" "bluetooth-hci-socket"
npm ERR! node v4.4.4
npm ERR! npm  v2.15.1
npm ERR! code ELIFECYCLE

npm ERR! bluetooth-hci-socket@0.5.0 install: `node-gyp rebuild`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the bluetooth-hci-socket@0.5.0 install script 'node-gyp rebuild'.
npm ERR! This is most likely a problem with the bluetooth-hci-socket package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-gyp rebuild
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs bluetooth-hci-socket
npm ERR! Or if that isn't available, you can get their info via:
npm ERR!
npm ERR!     npm owner ls bluetooth-hci-socket
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     C:\nuimo-electron\npm-debug.log
[npm-debug.txt](https://github.com/sandeepmistry/node-bluetooth-hci-socket/files/444417/npm-debug.txt)

```

sandeepmistry, 2016-09-05:
Hi @mrothenbuecher,

Unfortunately I'm not a full-time Windows user, so am not too sure what to suggest. My only suggestion is to follow's @don's Youtube guide: https://www.youtube.com/watch?v=mL9B8wuEdms

erwinkendo, 2016-09-09:
I think this is a problem with your windows and Visual Studio setup. If it's of any use, I compile with MSCV 2013 (Microsoft Visual Studio 12.0) and everything works fine.

TorstenRobitzki, 2016-09-09:
I have the very same issue with windows 7 and Visual Studio Community 2015 (same as in the youtube vid). I already tried the actual master of 'node-gyp' and 'node-bluetooth-hci-socket'.

sandeepmistry, 2016-09-09:
Has anyone tried this? https://stackoverflow.com/questions/38149603/npm-install-fails-with-error-c2373-with-vs2015-update-3

TorstenRobitzki, 2016-09-09:
I had to install 'rimraf' separately. But then this worked for me! :-) Thanks a lot!

sandeepmistry, 2016-09-11:
@TorstenRobitzki great to hear! Did you use the answer with the green checkmark? (http://stackoverflow.com/a/38149604/2020087)

Closing, as there is a know workaround and it seems to be an issue with npm.

closed 2016-09-11 by sandeepmistry

TorstenRobitzki, 2016-09-11:
yes, I've used the accepted answer.