noble/node-bluetooth-hci-socket #36: HRM disconection after 1 minute in Windows 8 machine

erwinkendo, 2016-08-31:
Hi @sandeepmistry

Thank you for your hard work, you have come a long way.

I started using noble and noble-device with Windows 8.1 just recently and discovered a problem when interacting with a hear rate monitor using the same example in the noble-devices repo. After 1 minute, or 60 measurements, the device gets disconnected by itself.

I already tried the same example on a Linux machine and the problem does not happen in Linux. I tried with two different ID 0a12:0001 Cambridge Silicon Radio, Ltd Bluetooth Dongle (HCI mode) dongles, but the result is the same.

Using the `NODE=*` tag, I managed to see that the last hci interaction before disconnecting is always the same in the windows machine (of course this is never experimented in the Linux Machine), 
```
called hci   event type = 4 +0ms
  hci   sub event type = 5 +1ms
  hci           handle = 70 +0ms
  hci           reason = 59 +0ms
```

Some data received before looks like this:

```
update measument: 70
  hci-usb ACL Data In: 46200d00090004001b120016468a034403 +1s
  hci onSocketData: 0246200d00090004001b120016468a034403 +0ms
  hci   event type = 2 +0ms
  hci           cid = 4 +1ms
  hci           handle = 70 +0ms
  hci           data = 1b120016468a034403 +0ms
update measument: 70
  hci-usb ACL Data In: 46200b00070004001b120016468b03 +1s
  hci onSocketData: 0246200b00070004001b120016468b03 +0ms
  hci   event type = 2 +0ms
  hci           cid = 4 +1ms
  hci           handle = 70 +0ms
  hci           data = 1b120016468b03 +0ms
update measument: 70
  hci-usb ACL Data In: 46200b00070004001b120016469e03 +1s
  hci onSocketData: 0246200b00070004001b120016469e03 +1ms
  hci   event type = 2 +0ms
  hci           cid = 4 +1ms
  hci           handle = 70 +0ms
  hci           data = 1b120016469e03 +0ms
update measument: 70
  hci-usb HCI event: 05040046003b +494ms
  hci onSocketData: 0405040046003b +1ms
  hci   event type = 4 +0ms
  hci   sub event type = 5 +1ms
  hci           handle = 70 +0ms
  hci           reason = 59 +0ms
disconnected!
  hci set scan enabled - writing: 010c20020001 +1ms
  hci-usb write: 010c20020001 +0ms
```

Any help would be greatly appreciated :)

sandeepmistry, 2016-09-05:
Hi @erwinkendo,

Could you please share the full debug log on Windows? As well as the HCI dump capture from Linux (`sudo hcidump -t -x`).

There might be a non-`4` CID type for connection interval that the Linux kernel is handling.

erwinkendo, 2016-09-10:
Hi

Sorry for the delay, here attached are both logs

I do not see anything different after 1 minute of operation (time when the disconnection happens), but hopefully you are better at noticing anything unusual.

Kind regards,
[hcidump.txt](https://github.com/sandeepmistry/node-bluetooth-hci-socket/files/465640/hcidump.txt)
[outDebugWin.txt](https://github.com/sandeepmistry/node-bluetooth-hci-socket/files/465641/outDebugWin.txt)

erwinkendo, 2016-09-19:
Just a small update:

Tried using the latest version of all packages, as well as using different adapters and USB ports but it does no seem to improve. Could this be related to the node version I am using? (4.5)

mrothenbuecher, 2016-09-20:
I have the same issue. 

Running node Version 6.6.0

erwinkendo, 2016-09-20:
Hi @mrothenbuecher , I am using noble-device example for [hrm connection](https://github.com/sandeepmistry/noble-device/blob/master/examples/hrm-device.js)

My last event is similar to yours... but the names change for some way:

```
update measument: 81
  hci-usb ACL Data In: 48200b00070004001b12001651f502 +990ms
  hci onSocketData: 0248200b00070004001b12001651f502 +1ms
  hci   event type = 2 +1ms
  hci           cid = 4 +1ms
  hci           handle = 72 +0ms
  hci           data = 1b12001651f502 +1ms
update measument: 81
  hci-usb HCI event: 05040048003b +478ms
  hci onSocketData: 0405040048003b +1ms
  hci   event type = 4 +1ms
  hci   sub event type = 5 +0ms
  hci           handle = 72 +1ms
  hci           reason = 59 +0ms
disconnected!
  hci set scan enabled - writing: 010c20020001 +3ms
  hci-usb write: 010c20020001 +1ms
```
Is it normal that the HCI event and other data is not the same for our systems @sandeepmistry ?

sandeepmistry, 2016-09-26:
The problem is nothing handles the following messages on Windows:

```
2016-09-10 21:21:44.061067 > ACL data: handle 33 flags 0x02 dlen 16
    L2CAP(d): cid 0x0005 len 12 [psm 0]
      12 39 08 00 FA 00 90 01 00 00 58 02 
```

The Linux kernel responds with:

```
2016-09-10 21:21:44.061134 < ACL data: handle 33 flags 0x00 dlen 10
    L2CAP(d): cid 0x0005 len 6 [psm 0]
      13 39 02 00 00 00 
2016-09-10 21:21:44.061140 < HCI Command: LE Connection Update (0x08|0x0013) plen 14
  21 00 FA 00 90 01 00 00 58 02 00 00 00 00 
2016-09-10 21:21:44.062661 > HCI Event: Command Status (0x0f) plen 4
    LE Connection Update (0x08|0x0013) status 0x00 ncmd 1
```

There is nothing in noble to handle this event and reply.

Could you please close this issue and open a new one in the noble Github repo?

erwinkendo, 2016-09-26:
Thanks @sandeepmistry 

Opened a new issue [here](https://github.com/sandeepmistry/noble/issues/468)

closed 2016-09-26 by erwinkendo