noble/node-bluetooth-hci-socket #49: TypeError: Cannot read property 'on' of undefined

hulkco, 2016-11-30:
Hi this error in my computer with Windows 10

C:\node\controlBLE>node control.js
C:\node\controlBLE\node_modules\noble\node_modules\bluetooth-hci-socket\lib\usb.js:98
    this._aclDataInEndpoint.on('data', this.onAclDataInEndpointData.bind(this));
                           ^

TypeError: Cannot read property 'on' of undefined
    at BluetoothHciSocket.start (C:\node\controlBLE\node_modules\noble\node_modules\bluetooth-hci-socket\lib\usb.js:98:28)
    at Hci.init (C:\node\controlBLE\node_modules\noble\lib\hci-socket\hci.js:100:18)                                        at NobleBindings.init (C:\node\controlBLE\node_modules\noble\lib\hci-socket\bindings.js:83:13)
    at new Noble (C:\node\controlBLE\node_modules\noble\lib\noble.js:50:18)                                                 at Object.<anonymous> (C:\node\controlBLE\node_modules\noble\index.js:4:18)
    at Module._compile (module.js:570:32)
    at Object.Module._extensions..js (module.js:579:10)
    at Module.load (module.js:487:32)
    at tryModuleLoad (module.js:446:12)
    at Function.Module._load (module.js:438:3)

The target device its Arduino 101: USB\VID_8087&PID_0AB6

sandeepmistry, 2016-12-04:
Hi @hulkco,

The Arduino 101 is not a compatible USB Bluetooth HCI device.

Please see the read me for devices that have been tested by others: https://github.com/sandeepmistry/node-bluetooth-hci-socket#compatible-bluetooth-40-usb-adapters

Closing for now, please re-open with more info. if there is a something I am misunderstanding.

closed 2016-12-04 by sandeepmistry

khushbooY, 2019-12-24:
Hi,
I guess your this line of code contains default endpoint addressess but there are some UsbDeviceInterface's Device endoints that have different address.

this._usbDevice.open();
this._usbDeviceInterface = this._usbDevice.interfaces[0];
this._aclDataOutEndpoint = this._usbDeviceInterface.endpoint(0x02);//like 0x01
this._hciEventEndpoint = this._usbDeviceInterface.endpoint(0x81);//like 0x83
this._aclDataInEndpoint = this._usbDeviceInterface.endpoint(0x82);//like 0x81
this._usbDeviceInterface.claim();

Firstly this showed error

Error from process TypeError: Cannot read property 'on' of undefined
at BluetoothHciSocket.start (D:\exe_Inprocess\bisoft_exe_64\node_modules_x64\node_modules\bluetooth-hci-socket\lib\usb.js:99:29)
at Hci.init (D:\exe_Inprocess\bisoft_exe_64\node_modules_x64\node_modules\noble\lib\hci-socket\hci.js:102:18)
at NobleBindings.init (D:\exe_Inprocess\bisoft_exe_64\node_modules_x64\node_modules\noble\lib\hci-socket\bindings.js:82:13)
at Noble. (D:\exe_Inprocess\bisoft_exe_64\node_modules_x64\node_modules\noble\lib\noble.js:57:24)
at _combinedTickCallback (internal/process/next_tick.js:131:7)
at process._tickCallback (internal/process/next_tick.js:180:9)
then i changed it to the UsbDeviceInterface's Device endoints then only scanning of devices is not happening?

Could you please help???