noble/node-bluetooth-hci-socket #50: using userBind cause error on setFilter

slashmili, 2016-12-02:
Not sure if it's a bug or not having proper configuration, I tried to change `le-scan-test.js` to use `bluetoothHciSocket.bindUser()` and when run it I get `[Error: File descriptor in bad state]`. It's happens when code sets in `bluetoothHciSocket.setFilter`. BTW beside that error,  looks like scanning works

this the change:
```
diff --git a/examples/le-scan-test.js b/examples/le-scan-test.js
index e573c53..8912a5d 100644
--- a/examples/le-scan-test.js
+++ b/examples/le-scan-test.js
@@ -120,7 +120,7 @@ function setScanEnable(enabled, duplicates) {
   bluetoothHciSocket.write(cmd);
 }

-bluetoothHciSocket.bindRaw();
+bluetoothHciSocket.bindUser();
 setFilter();
 bluetoothHciSocket.start();
```

and this is the output
```
sudo hciconfig hci0 down
sudo node le-scan-test.js
[Error: File descriptor in bad state]
data: 040e04010c200c
data: 040e04010b2000
LE Scan Parameters Set
data: 040e04010c2000
LE Scan Enable Set
data: 043e1702010000efe1d4c2153c0b02010607ff4c0010020b00c7
LE Advertising Report
	ADV_IND
	PUBLIC
	3c:15:c2:d4:e1:ef
	02010607ff4c0010020b00
	-57
data: 043e0c02010400efe1d4c2153c00c6
LE Advertising Report
	SCAN_RSP
	PUBLIC
	3c:15:c2:d4:e1:ef

	-58
data: 043e2b02010000ca09cc6ebb141f02011e1bff7500420401404f14bb6ecc09ca16bb6e5109c908000000000000ae
```

sandeepmistry, 2016-12-05:
@slashmili please see: https://github.com/sandeepmistry/node-bluetooth-hci-socket#user-channel

For user channel the adapter must be powered down before starting the Node.js app as per the documentation.

slashmili, 2016-12-05:
@sandeepmistry as I mentioned in the output I already powered down the device
```
sudo hciconfig hci0 down
sudo node le-scan-test.js
...
```

sandeepmistry, 2016-12-05:
@slashmili apologies for missing this.

I believe `setFilter` is only applicable for `bindRaw` and not `bindUser` mode, as you are still getting an the events.

Could you please submit a PR to read me to document this?

slashmili, 2016-12-05:
Sure!

Does it mean that in [noble/hci.js](https://github.com/sandeepmistry/noble/blob/97cf4414fee88d1f29ee7d6aeb2f35fc0613e606/lib/hci-socket/hci.js#L113), there should be a checking to only call `setSocketFilter` if `HCI_CHANNEL_RAW` is set?

The reason I'm interested to these details is, I'm porting this library to `Erlang` and planning to use it in [Nerves](http://nerves-project.org/)

sandeepmistry, 2016-12-08:
> Does it mean that in noble/hci.js, there should be a checking to only call setSocketFilter if HCI_CHANNEL_RAW is set?

Yes, I agree. It currently ignores the error event of "File descriptor in bad state" - but it would be clearer to not call `setSocketFilter` as you suggested. Do you have time to submit a PR for this as well?

slashmili, 2016-12-08:
Of course! I'll do this this week

sandeepmistry, 2017-01-16:
@slashmili any progress on the PR?

slashmili, 2017-01-23:
@sandeepmistry no sorry I'm still pretty busy moving 🚚

slashmili, 2017-02-15:
Just for the record I found this explanation  

> Using this new facility does not require any external library or special
> **ioctl or socket filters.** Just create the socket and bind it. After that
> the file descriptor is ready to speak H:4 protocol.
> 

http://www.spinics.net/lists/linux-bluetooth/msg37345.html

closed 2017-02-15 by slashmili