noble/node-bluetooth-hci-socket #51: Add support for Intel(R) Wireless Bluetooth(R) PID 07DA

aj-ptw, 2016-12-07:
@sandeepmistry we @openbci have integrated your noble and were wondering what is the process for adding different devices to this repo?

The VID is the same as the other Intel chip supported `0x8087` however the PID is a bit different of `0x07DA` (the supported one is `0x07DC`)

I tried adding the PID and VID to the whitelist in [usb.js](../blob/master/lib/usb.js) but was not successful in getting it to work. Getting a `LIBUSB_NOT_SUPPORTED` error when trying to run the [scan example](../blob/master/examples/le-scan-test.js)

ksantos, 2016-12-20:
@aj-ptw
Im trying the same, my PID is 0x0a2a and Im getting the same error LIBUSB_NOT_SUPPORTED. There is no a workaround to solve this?

rsahores, 2017-01-08:
@ksantos I've added PID 0x0a2a in usb.js, switched driver to WinUSB with Zadig tool and it worked perfectly.

sandeepmistry, 2017-01-16:
Just a PR like https://github.com/sandeepmistry/node-bluetooth-hci-socket/pull/53 is needed.

Make sure the WINUSB driver is installed for the adapter as per the read me: https://github.com/sandeepmistry/node-bluetooth-hci-socket#windows

closed 2017-01-16 by sandeepmistry