noble/node-bluetooth-hci-socket #55: Extending USB Driver Compatibilty

navadiatarun, 2017-02-06:
Wanted to know that what does this code snippet represent inside https://github.com/sandeepmistry/node-bluetooth-hci-socket/blob/master/lib/usb.js:
`
this._usbDevice.open();

  this._usbDeviceInterface = this._usbDevice.interfaces[0];

  this._aclDataOutEndpoint = this._usbDeviceInterface.endpoint(0x02);

  this._hciEventEndpoint = this._usbDeviceInterface.endpoint(0x81);
  this._aclDataInEndpoint = this._usbDeviceInterface.endpoint(0x82);

  this._usbDeviceInterface.claim();
};
`
I want to make your library compatible with this Device 
Marvell AVASTAR Wireless Composite Device driver
: USB VID 1286
PID :204C

Can we change some of the parameters in the above snippet, so that the binding is successful.

sandeepmistry, 2017-02-07:
@navadiatarun you'll just need a simple PR like https://github.com/sandeepmistry/node-bluetooth-hci-socket/pull/40

navadiatarun, 2017-02-07:
@sandeepmistry I added the Pid and Vid in environment variables, that throws a different error.
`:C:\Users\Intel\Desktop\Workspace\node_modules\bluetooth-hci-socket\lib\usb.js:95 this._hciEventEndpoint.on('data', this.onHciEventEndpointData.bind(this)); ^
TypeError: Cannot read property 'on' of undefined at BluetoothHciSocket.start (C:\Users\Intel\Desktop\Workspace\node_modules\bluetooth-hci-socket\lib\usb.js:95:27) at Hci.init (C:\Users\Intel\Desktop\Workspace\node_modules\noble\lib\hci-socket\hci.js:100:18) at NobleBindings.init (C:\Users\Intel\Desktop\Workspace\node_modules\noble\lib\hci-socket\bindings.js:83:13) at new Noble (C:\Users\Intel\Desktop\Workspace\node_modules\noble\lib\noble.js:50:18) at Object. (C:\Users\Intel\Desktop\Workspace\node_modules\noble\index.js:4:18) at Module.compile (module.js:570:32) at Object.Module._extensions..js (module.js:579:10) at Module.load (module.js:487:32) at tryModuleLoad (module.js:446:12) at Function.Module.load (module.js:438:3)`

After changing the endpoint addresses in the below code snippet 
`
https://github.com/sandeepmistry/node-bluetooth-hci-socket/blob/master/lib/usb.js: this._usbDevice.open();
this.usbDeviceInterface = this.usbDevice.interfaces[0];
this._aclDataOutEndpoint = this._usbDeviceInterface.endpoint(0x04);
this.hciEventEndpoint = this._usbDeviceInterface.endpoint(0x82); this.aclDataInEndpoint = this._usbDeviceInterface.endpoint(0x83);
this._usbDeviceInterface.claim(); }; 
`

 The code is able to scan for classic Bluetooth but not for BLE, as far as I know, the Marvell AVASTAR Wireless Composite Device driver is BLE compatible and it works through the Windows Bluetooth UI.

sandeepmistry, 2017-02-13:
> The code is able to scan for classic Bluetooth but not for BLE, as far as I know, the Marvell AVASTAR Wireless Composite Device driver is BLE compatible and it works through the Windows Bluetooth UI.

What make you say that? What code are you running? Have you tried debug logs?

```
setenv DEBUG=*
node <file>.js
```

navadiatarun, 2017-04-11:
Tried changing the settings here
this._usbDeviceInterface = this._usbDevice.interfaces[0];

this._aclDataOutEndpoint = this._usbDeviceInterface.endpoint(0x02);

this._hciEventEndpoint = this._usbDeviceInterface.endpoint(0x81);
this._aclDataInEndpoint = this._usbDeviceInterface.endpoint(0x82);

this._usbDeviceInterface.claim();
};

This is not possible, hence closing the issue.

closed 2017-04-11 by navadiatarun