noble/node-bluetooth-hci-socket #57: Unable to install on Raspberry Pi 3

IgorGanapolsky, 2017-03-03:
Hi,
I am unable to install your package on Raspberry Pi 3.  Here is the error:
> ostuninstall bluetooth-hci-socket@0.5.1
2731 error bluetooth-hci-socket@0.5.1 install: `node-gyp rebuild`
2731 error Exit status 1
2732 error Failed at the bluetooth-hci-socket@0.5.1 install script.
2732 error This is most likely a problem with the bluetooth-hci-socket package,
2732 error not with npm itself.
2732 error Tell the author that this fails on your system:
2732 error     node-gyp rebuild
2732 error You can get their info via:
2732 error     npm owner ls bluetooth-hci-socket
2732 error There is likely additional logging output above.

sandeepmistry, 2017-03-13:
@IgorGanapolsky any more details from the console output, there's not really much we can help you out without more details.

Please provide mode details and re-open this issue.

closed 2017-03-13 by sandeepmistry

IgorGanapolsky, 2017-03-13:
@sandeepmistry I was able to resolve it by `npm install -g node-gyp`