noble/node-bluetooth-hci-socket #58: rename node-pre-gyp module_name

farfromrefug, 2017-03-07:
Right now the module_name for node-pre-gyp is "binding"
However if you want to provide your [own custom hosted binaries](https://github.com/mapbox/node-pre-gyp#download-binary-files-from-a-mirror), it depends on "module_name". Thus "module_name" should be something like "bluetooth-hci-socket"

That would be awesome!

sandeepmistry, 2017-03-13:
@farfromrefug please submit a pull request for this.

jrobeson, 2018-06-18:
a PR that includes that change is here https://github.com/noble/node-bluetooth-hci-socket/pull/87

jrobeson, 2018-06-18:
Oh sorry, I totally missed the PR already submitted by you!

rzr, 2020-10-16:
Is this issue relevant in this release:

https://www.npmjs.com/package/@abandonware/bluetooth-hci-socket/v/0.5.3-6

rzr, 2021-01-06:
Feel free to comment this related ticket:

https://github.com/abandonware/node-bluetooth-hci-socket/issues/9

rzr, 2021-10-01:
May you forward your ticket to downstream version:

https://github.com/noble/node-bluetooth-hci-socket/issues/123