noble/node-bluetooth-hci-socket #64: On restart the serial bus driver needs re-install

T-Rez-Mity, 2017-04-20:
>`Microsoft Windows [Version 10.0.10586]
>(c) 2015 Microsoft Corporation. All rights reserved.
>
>C:\Users\Trezor.Owens>node-red
>Welcome to Node-RED
>===================
> 20 Apr 12:13:46 - [info] Node-RED version: v0.15.1
> 20 Apr 12:13:46 - [info] Node.js  version: v6.10.2
> 20 Apr 12:13:46 - [info] Windows_NT 10.0.10586 x64 LE
> 20 Apr 12:13:47 - [info] Loading palette nodes
> 20 Apr 12:13:49 - [warn] ------------------------------------------------------
> 20 Apr 12:13:49 - [warn] [rpi-gpio] Info : Ignoring Raspberry Pi specific node
> 20 Apr 12:13:49 - [warn] [tail] Not currently supported on Windows.
> 20 Apr 12:13:49 - [warn] [serialport] Error: Module version mismatch. Expected 4
> 8, got 46.
> 20 Apr 12:13:49 - [warn] ------------------------------------------------------
> 20 Apr 12:13:49 - [info] Settings file  : C:\Users\Trezor.Owens\AppData\Roaming\
> npm\node_modules\node-red\settings.js
> 20 Apr 12:13:49 - [info] User directory : \Users\Trezor.Owens\.node-red
> 20 Apr 12:13:49 - [info] Flows file     : \Users\Trezor.Owens\.node-red\flows_00
> 1814-LAP.json
> 20 Apr 12:13:49 - [info] Server now running at http://127.0.0.1:1880/
> 20 Apr 12:13:49 - [info] Starting flows
> 20 Apr 12:13:49 - [info] Started flows
> 20 Apr 12:13:49 - [red] Uncaught Exception:
> 20 Apr 12:13:49 - RangeError: Index out of range
>     at checkOffset (buffer.js:834:11)
>     at Buffer.readUInt8 (buffer.js:872:5)
>     at Hci.processCmdCompleteEvent (C:\Users\Trezor.Owens\.node-red\node_modules
> \noble\lib\hci-socket\hci.js:566:25)
>     at Hci.onSocketData (C:\Users\Trezor.Owens\.node-red\node_modules\noble\lib\
> hci-socket\hci.js:461:12)
>     at emitOne (events.js:96:13)
>     at BluetoothHciSocket.emit (events.js:188:7)
>     at BluetoothHciSocket.onHciEventEndpointData (C:\Users\Trezor.Owens\.node-re
> d\node_modules\noble\node_modules\bluetooth-hci-socket\lib\usb.js:156:10)
>     at emitOne (events.js:96:13)
>     at InEndpoint.emit (events.js:188:7)
>     at Transfer.transferDone (C:\Users\Trezor.Owens\.node-red\node_modules\usb\u
> sb.js:314:9)'
> 

Here I update the usb serial bus device drive back to the default intel bluetooth driver, then re-install using zadig and it works as below.  Any thoughts?

> C:\Users\Trezor.Owens>node-red
> 
> 
> Welcome to Node-RED
> ===================
> 
> 20 Apr 12:14:41 - [info] Node-RED version: v0.15.1
> 20 Apr 12:14:41 - [info] Node.js  version: v6.10.2
> 20 Apr 12:14:41 - [info] Windows_NT 10.0.10586 x64 LE
> 20 Apr 12:14:42 - [info] Loading palette nodes
> 20 Apr 12:14:43 - [warn] ------------------------------------------------------
> 20 Apr 12:14:43 - [warn] [rpi-gpio] Info : Ignoring Raspberry Pi specific node
> 20 Apr 12:14:43 - [warn] [tail] Not currently supported on Windows.
> 20 Apr 12:14:43 - [warn] [serialport] Error: Module version mismatch. Expected 4
> 8, got 46.
> 20 Apr 12:14:43 - [warn] ------------------------------------------------------
> 20 Apr 12:14:43 - [info] Settings file  : C:\Users\Trezor.Owens\AppData\Roaming\
> npm\node_modules\node-red\settings.js
> 20 Apr 12:14:43 - [info] User directory : \Users\Trezor.Owens\.node-red
> 20 Apr 12:14:43 - [info] Flows file     : \Users\Trezor.Owens\.node-red\flows_00
> 1814-LAP.json
> 20 Apr 12:14:43 - [info] Server now running at http://127.0.0.1:1880/
> 20 Apr 12:14:43 - [info] Starting flows
> 20 Apr 12:14:43 - [info] Started flows

sandeepmistry, 2017-04-23:
@T-Rez-Mity is there anyway to reproduce this without using node-red? What BLE devices were you using?

T-Rez-Mity, 2017-04-24:
Sensortag CC2650. I will try without node-red.

sandeepmistry, 2017-05-08:
@T-Rez-Mity any other BLE devices in range that node-red is setup for?

What happens if you set the `DEBUG` env. var to `hci-usb` before starting `node-red`. I think it's something like `setenv DEBUG=hci-usb`.

rzr, 2020-10-16:
Is this issue also relevant for this release:

https://www.npmjs.com/package/@abandonware/bluetooth-hci-socket/v/0.5.3-6