noble/node-bluetooth-hci-socket #65: Interfacing two bluetooth modules with the rpi

sb-mukul23, 2017-04-21:
I am trying to interface two bluetooth modules with the rpi. One would act as a peripheral and the other would be detected as an ibeacon. But it is just behaving as a peripheral and I not able to detect the second module

sandeepmistry, 2017-04-23:
@sb-mukul23 you have to pass in the device id (0, 1, etc.) into the `bind*` methods: https://github.com/sandeepmistry/node-bluetooth-hci-socket#bind

closed 2017-04-23 by sandeepmistry

sb-mukul23, 2017-04-24:
thanks alot, but I could not find the bind methods

sb-mukul23, 2017-04-24:
var bleno = require('bleno');

var SystemInformationService = require('./systeminformationservice');
var BluetoothHciSocket = require('bluetooth-hci-socket');

var systemInformationService = new SystemInformationService();

var bluetoothHciSocket = new BluetoothHciSocket();

bluetoothHciSocket.bindRaw(6);
bluetoothHciSocket.bindUser(5);

bleno.on('stateChange', function(state) {
  console.log('on -> stateChange: ' + state);

  if (state === 'poweredOn') {

    bleno.startAdvertising(bleno.name, [systemInformationService.uuid]);
  }
  else {

    bleno.stopAdvertising();
  }
});

bleno.on('advertisingStart', function(error) {

  console.log('on -> advertisingStart: ' +
    (error ? 'error ' + error : 'success')
  );

  if (!error) {

    bleno.setServices([
      systemInformationService
    ]);
  }
});

this is my index.js file. My bluetooth modules have device id 5 and 6 on the rpi and I am using lightblue which is an IOS app to detect ble. I can only find the rpi and cannot find the second bluetooth module. Also I wanted to know if I could make my rpi as the master and send commands from rpi to the phone

sandeepmistry, 2017-05-08:
See: https://github.com/sandeepmistry/noble#multiple-adapters

and

https://github.com/sandeepmistry/bleno#multiple-adapters