noble/node-bluetooth-hci-socket #79: error on bluetooth-hci-socket@0.5.1 install

lewildfox, 2018-03-07:
Hello,

I'm totally new to node & nodered.
I'm trying to install "node-red-contrib-sensortag" or "node-red-node-sensortag" to connect TI sensor tag to WatsonIoT, but encountered the below error.
I'm using raspbian Jessie on raspberry pi 3B, with below module versions:
node -v
v8.10.0
nodejs -v
v8.10.0
npm -v
3.10.10

What should I do?
Appreciate any clue and help. :)

Thank you!
Adi

`pi@raspberrypi:~/.node-red $ npm install node-red-contrib-sensortag

> usb@1.3.1 install /home/pi/.node-red/node_modules/usb
> node-pre-gyp install --fallback-to-build

node-pre-gyp ERR! Tried to download(404): https://github.com/tessel/node-usb/releases/download/1.3.1/usb_bindings-v1.3.1-node-v57-linux-arm.tar.gz 
node-pre-gyp ERR! Pre-built binaries not found for usb@1.3.1 and node@8.10.0 (node-v57 ABI, glibc) (falling back to source compile with node-gyp) 
make: Entering directory '/home/pi/.node-red/node_modules/usb/build'
  CC(target) Release/obj.target/libusb/libusb/libusb/core.o
  CC(target) Release/obj.target/libusb/libusb/libusb/descriptor.o
  CC(target) Release/obj.target/libusb/libusb/libusb/hotplug.o
  CC(target) Release/obj.target/libusb/libusb/libusb/io.o
  CC(target) Release/obj.target/libusb/libusb/libusb/strerror.o
  CC(target) Release/obj.target/libusb/libusb/libusb/sync.o
  CC(target) Release/obj.target/libusb/libusb/libusb/os/poll_posix.o
  CC(target) Release/obj.target/libusb/libusb/libusb/os/threads_posix.o
  CC(target) Release/obj.target/libusb/libusb/libusb/os/linux_usbfs.o
  CC(target) Release/obj.target/libusb/libusb/libusb/os/linux_udev.o
  AR(target) Release/obj.target/usb.a
  COPY Release/usb.a
  CXX(target) Release/obj.target/usb_bindings/src/node_usb.o
  CXX(target) Release/obj.target/usb_bindings/src/device.o
  CXX(target) Release/obj.target/usb_bindings/src/transfer.o
  SOLINK_MODULE(target) Release/obj.target/usb_bindings.node
  COPY Release/usb_bindings.node
  COPY /home/pi/.node-red/node_modules/usb/src/binding/usb_bindings.node
  TOUCH Release/obj.target/action_after_build.stamp
make: Leaving directory '/home/pi/.node-red/node_modules/usb/build'

> bluetooth-hci-socket@0.5.1 install /home/pi/.node-red/node_modules/bluetooth-hci-socket
> node-gyp rebuild

**make: Entering directory '/home/pi/.node-red/node_modules/bluetooth-hci-socket/build'
  CXX(target) Release/obj.target/binding/src/BluetoothHciSocket.o
../src/BluetoothHciSocket.cpp: In member function ‘void BluetoothHciSocket::emitErrnoError()’:
../src/BluetoothHciSocket.cpp:275:72: warning: ‘v8::Local<v8::Object> v8::Function::NewInstance(int, v8::Local<v8::Value>*) const’ is deprecated (declared at /home/pi/.node-gyp/8.10.0/include/node/v8.h:3846): Use maybe version [-Wdeprecated-declarations]
   Local<Value> error = errorConstructor->NewInstance(1, constructorArgs);**
                                                                        ^
  SOLINK_MODULE(target) Release/obj.target/binding.node
  COPY Release/binding.node
make: Leaving directory '/home/pi/.node-red/node_modules/bluetooth-hci-socket/build'
node-red-project@0.0.1 /home/pi/.node-red

`

sandeepmistry, 2018-03-12:
Hi @lewildfox,

Looks like the install went well, there is just a warning about an API deprecation for Node 8.

closed 2018-03-12 by sandeepmistry

lewildfox, 2018-03-18:
Hi sandeep,

Okay, thanks anyway for your confirmation.
Cheers!