noble/node-bluetooth-hci-socket #84: Cannot install on node 10

kounch, 2018-05-06:
`
(...)
In file included from ../src/BluetoothHciSocket.cpp:8:0:
../../nan/nan.h:950:46: nota: se declara aquí
   NAN_DEPRECATED inline v8::Local<v8::Value> MakeCallback(
                                              ^~~~~~~~~~~~
../src/BluetoothHciSocket.cpp: En la función miembro void BluetoothHciSocket::emitErrnoError():
../src/BluetoothHciSocket.cpp:275:72: error: no matching function for call to v8::Function::NewInstance(int, v8::Local<v8::Value> [1])
   Local<Value> error = errorConstructor->NewInstance(1, constructorArgs);
(...)
`

Possibly related to old nan dependency, like is mentioned here:

https://github.com/agnat/node_mdns/pull/210

mogenson, 2018-05-07:
On Linux x86_64 with nodejs 10.0.0-1

```
In file included from ../src/BluetoothHciSocket.cpp:8:0:
../../nan/nan.h:950:46: note: declared here
   NAN_DEPRECATED inline v8::Local<v8::Value> MakeCallback(
                                              ^~~~~~~~~~~~
make: *** [binding.target.mk:95: Release/obj.target/binding/src/BluetoothHciSocket.o] Error 1
make: Leaving directory '/path/to/my/node/project/node_modules/bluetooth-hci-socket/build'
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:258:23)
gyp ERR! stack     at ChildProcess.emit (events.js:182:13)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:225:12)
gyp ERR! System Linux 4.16.7-1-MANJARO
gyp ERR! command "/usr/bin/node" "/usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /home/mike/Scansorial/root-bootloader/xsquare/node_modules/bluetooth-hci-socket
gyp ERR! node -v v10.0.0
gyp ERR! node-gyp -v v3.6.2
gyp ERR! not ok 
```

kounch, 2018-05-25:
After some tests, it looks like making these two changes solve the problem:

File  **src/BluetoothHciSocket.cpp**, change line 276 from
`    Local<Value> error = errorConstructor->NewInstance(1, constructorArgs);`
to
`     Local<Value> error = Nan::NewInstance(errorConstructor, 1, constructorArgs).ToLocalChecked();`

File **package.json**, change line 30 from
`    "nan": "^2.0.5",`
to
`    "nan": "^2.10.0",`

airhorns, 2018-06-01:
I'm also getting this error and can confirm downgrading to node `8.11.1` fixed my issue. My full build log:

```
[main]  info This module is OPTIONAL, you can safely ignore this error
[main]  warning Error running install script for optional dependency: "/node_modules/bluetooth-hci-socket: Command failed.
[main]  Exit code: 1
[main]  Command: node-gyp rebuild
[main]  Arguments:
[main]  Directory: /node_modules/bluetooth-hci-socket
[main]  Output:
[main]  gyp info it worked if it ends with ok
[main]  gyp info using node-gyp@3.6.2
[main]  gyp info using node@10.0.0 | linux | arm
[main]  gyp info spawn /usr/bin/python2
[main]  gyp info spawn args [ '/usr/local/lib/node_modules/npm/node_modules/node-gyp/gyp/gyp_main.py',
[main]  gyp info spawn args   'binding.gyp',
[main]  gyp info spawn args   '-f',
[main]  gyp info spawn args   'make',
[main]  gyp info spawn args   '-I',
[main]  gyp info spawn args   '/node_modules/bluetooth-hci-socket/build/config.gypi',
[main]  gyp info spawn args   '-I',
[main]  gyp info spawn args   '/usr/local/lib/node_modules/npm/node_modules/node-gyp/addon.gypi',
[main]  gyp info spawn args   '-I',
[main]  gyp info spawn args   '/root/.node-gyp/10.0.0/include/node/common.gypi',
[main]  gyp info spawn args   '-Dlibrary=shared_library',
[main]  gyp info spawn args   '-Dvisibility=default',
[main]  gyp info spawn args   '-Dnode_root_dir=/root/.node-gyp/10.0.0',
[main]  gyp info spawn args   '-Dnode_gyp_dir=/usr/local/lib/node_modules/npm/node_modules/node-gyp',
[main]  gyp info spawn args   '-Dnode_lib_file=/root/.node-gyp/10.0.0/<(target_arch)/node.lib',
[main]  gyp info spawn args   '-Dmodule_root_dir=/node_modules/bluetooth-hci-socket',
[main]  gyp info spawn args   '-Dnode_engine=v8',
[main]  gyp info spawn args   '--depth=.',
[main]  gyp info spawn args   '--no-parallel',
[main]  gyp info spawn args   '--generator-output',
[main]  gyp info spawn args   'build',
[main]  gyp info spawn args   '-Goutput_dir=.' ]
[main]  gyp info spawn make
[main]  gyp info spawn args [ 'BUILDTYPE=Release', '-C', 'build' ]
[main]  make: Entering directory '/node_modules/bluetooth-hci-socket/build'
[main]    CXX(target) Release/obj.target/binding/src/BluetoothHciSocket.o
[main]  ../src/BluetoothHciSocket.cpp: In member function ‘void BluetoothHciSocket::poll()’:
[main]  ../src/BluetoothHciSocket.cpp:251:95: warning: ‘v8::Local<v8::Value> Nan::MakeCallback(v8::Local<v8::Object>, v8::Local<v8::String>, int, v8::Local<v8::Value>*)’ is deprecated (declared at ../../nan/nan.h:950) [-Wdeprecated-declarations]
[main]       Nan::MakeCallback(Nan::New<Object>(this->This), Nan::New(\"emit\").ToLocalChecked(), 2, argv);
[main]                                                                                                 ^
[main]  ../src/BluetoothHciSocket.cpp: In member function ‘void BluetoothHciSocket::emitErrnoError()’:
[main]  ../src/BluetoothHciSocket.cpp:275:72: error: no matching function for call to ‘v8::Function::NewInstance(int, v8::Local<v8::Value> [1])’
[main]     Local<Value> error = errorConstructor->NewInstance(1, constructorArgs);
[main]                                                                          ^
[main]  ../src/BluetoothHciSocket.cpp:275:72: note: candidates are:
[main]  In file included from /root/.node-gyp/10.0.0/include/node/node.h:63:0,
[main]                   from /root/.node-gyp/10.0.0/include/node/node_buffer.h:25,
[main]                   from ../src/BluetoothHciSocket.cpp:7:
[main]  /root/.node-gyp/10.0.0/include/node/v8.h:3848:44: note: v8::MaybeLocal<v8::Object> v8::Function::NewInstance(v8::Local<v8::Context>, int, v8::Local<v8::Value>*) const
[main]     V8_WARN_UNUSED_RESULT MaybeLocal<Object> NewInstance(
[main]                                              ^
[main]  /root/.node-gyp/10.0.0/include/node/v8.h:3848:44: note:   candidate expects 3 arguments, 2 provided
[main]  /root/.node-gyp/10.0.0/include/node/v8.h:3851:44: note: v8::MaybeLocal<v8::Object> v8::Function::NewInstance(v8::Local<v8::Context>) const
[main]     V8_WARN_UNUSED_RESULT MaybeLocal<Object> NewInstance(
[main]                                              ^
[main]  /root/.node-gyp/10.0.0/include/node/v8.h:3851:44: note:   candidate expects 1 argument, 2 provided
[main]  ../src/BluetoothHciSocket.cpp:282:93: warning: ‘v8::Local<v8::Value> Nan::MakeCallback(v8::Local<v8::Object>, v8::Local<v8::String>, int, v8::Local<v8::Value>*)’ is deprecated (declared at ../../nan/nan.h:950) [-Wdeprecated-declarations]
[main]     Nan::MakeCallback(Nan::New<Object>(this->This), Nan::New(\"emit\").ToLocalChecked(), 2, argv);
[main]                                                                                               ^
[main]  binding.target.mk:93: recipe for target 'Release/obj.target/binding/src/BluetoothHciSocket.o' failed
[main]  make: *** [Release/obj.target/binding/src/BluetoothHciSocket.o] Error 1
[main]  make: Leaving directory '/node_modules/bluetooth-hci-socket/build'
[main]  gyp ERR! build error
[main]  gyp ERR! stack Error: `make` failed with exit code: 2
[main]  gyp ERR! stack     at ChildProcess.onExit (/usr/local/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:258:23)
[main]  gyp ERR! stack     at ChildProcess.emit (events.js:182:13)
[main]  gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:225:12)
[main]  gyp ERR! System Linux 4.12.13
[main]  gyp ERR! command \"/usr/local/bin/node\" \"/usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js\" \"rebuild\"
[main]  gyp ERR! cwd /node_modules/bluetooth-hci-socket
[main]  gyp ERR! node -v v10.0.0
[main]  gyp ERR! node-gyp -v v3.6.2
[main]  gyp ERR! not ok"
```

jrobeson, 2018-06-18:
@kounch : I submitted what you wrote as PR here: https://github.com/noble/node-bluetooth-hci-socket/pull/91 , but you should probably submit it yourself. Once you do so, i'll delete this one.

rzr, 2018-11-08:
Any progress on this ?
Now node-10 is LTS so I guess it is more than welcome.

Current workaround is:

    npm install "https://github.com/jrobeson/node-bluetooth-hci-socket/#fix-builds-for-node-10" # worked

What is blocking ? I can help on it if needed...

Relate-to: https://github.com/mozilla-iot/gateway/pull/1441

Apollon77, 2018-11-26:
Any news here? it is very important for all projects that care about full nodejs support or to use current versions! What is so problematic here?

jrobeson, 2018-11-26:
there's nothing problematic. the maintainer is just not around.

jrobeson, 2018-11-26:
I've considered forking it, but I haven't wanted to maintain it alone.

AlCalzone, 2018-11-26:
> there's nothing problematic. the maintainer is just not around.

That does sound kinda problematic ;)

rzr, 2018-11-26:
Do you know existing github/npm team that could do co maintenance on this module ?

If none may I suggest to move this project to:
https://github.com/abandonware

jrobeson, 2018-11-26:
@AlCalzone : i meant there's nothing problematic with the PR itself :)

barcating, 2018-11-29:
I'm also getting this error.
/home/barcating/.node-gyp/11.3.0/include/node/v8config.h:326:3: note: in definition of macro ‘V8_DEPRECATED’
   declarator __attribute__((deprecated(message)))
   ^
binding.target.mk:101: recipe for target 'Release/obj.target/binding/src/BluetoothHciSocket.o' failed
make: *** [Release/obj.target/binding/src/BluetoothHciSocket.o] Error 1
make: Leaving directory '/home/barcating/SDR/bluetooth/gattacker/node_modules/bluetooth-hci-socket/build'
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:262:23)
gyp ERR! stack     at ChildProcess.emit (events.js:182:13)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:254:12)
gyp ERR! System Linux 4.13.0-36-generic
gyp ERR! command "/usr/bin/node" "/usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /home/barcating/SDR/bluetooth/gattacker/node_modules/bluetooth-hci-socket
gyp ERR! node -v v11.3.0
gyp ERR! node-gyp -v v3.8.0
gyp ERR! not ok

barcating, 2018-11-29:
> Any progress on this ?
> Now node-10 is LTS so I guess it is more than welcome.
> 
> Current workaround is:
> 
> ```
> npm install "https://github.com/jrobeson/node-bluetooth-hci-socket/#fix-builds-for-node-10" # worked
> ```
> What is blocking ? I can help on it if needed...
> 
> Relate-to: [mozilla-iot/gateway#1441](https://github.com/mozilla-iot/gateway/pull/1441)

Thanks！It worked.

rzr, 2018-11-29:
So please vote for:
https://github.com/noble/node-bluetooth-hci-socket/pull/91

rzr, 2018-12-14:
I forked it in:
https://github.com/abandonware/node-bluetooth-hci-socket
if anyone want to co maintain it let me know to and join the team

Then it could be released in NPM's org subrepo ?

Apollon77, 2018-12-14:
@rzr great idea! Thank you for taking the idea over! For NPM release I have no idea whats best.

BTW: In theory wealso would need to fork noble the same way because it is used there too and many users are directly using noble ... What do you think?

Apollon77, 2018-12-14:
I would need to be in the project to activate appveyor :-( Then you need to ssign me rights/invite me.
ELse Appveyor is also easy ... go https://appveyor.com and Login with Github and Add Project :-)

rzr, 2018-12-14:
I invited you to join
https://github.com/orgs/abandonware/people

Apollon77, 2018-12-14:
Got nothing so far and link shows only that I can not see anything :-)

Apollon77, 2018-12-17:
Can you check invite again? then I can activate apveyor too.
And BTW What do we do with noble? In theory we also need to fork this to use the new npm package in there :-(

rzr, 2018-12-17:
I invited you again in:
https://github.com/orgs/abandonware/people

Then I plan to upload npm in this org:
https://www.npmjs.com/org/abandonware

Unless upstream shows up and do maintenance job

Please track:
https://github.com/noble/node-bluetooth-hci-socket/issues/85

Apollon77, 2018-12-17:
Worked now but I also need repo rights to activate appveyor :-(

Apollon77, 2018-12-18:
I would love to support here, but c++ is not really one of my strengths ;-)
@rzr @jrobeson What about setting up a slack or such for direct exchange on topic?

rzr, 2018-12-18:
I am on IRC but I think project is using gitter

Feel free to test raise issue on tracker too:
https://www.npmjs.com/package/@abandonware/bluetooth-hci-socket

Meanwhile I'll track:
https://github.com/noble/node-bluetooth-hci-socket/issues/85

th317erd, 2018-12-22:
I wouldn't mind helping in my spare time with the C++ aspect of this. Hit me up

rzr, 2018-12-22:
Then please test:
https://www.npmjs.com/org/abandonware
And report success on:
https://github.com/abandonware/noble/issues/2

jrobeson, 2018-12-22:
@th317erd : are you familiar with node's n-api? It'd be nice to convert the addon to n-api so the module will work across different node versions.  We could go with plain C via n-api itself, or C++ via node-addon-api.

th317erd, 2018-12-23:
No, I am not familiar with n-api, or V8 interfaces. However, I am good with C/C++, which is why I offered. I will take a look @rzr and @jrobeson when I get a free moment. @rzr was there anything specific you wanted me to test?

janhaa, 2019-09-16:
This was just merged in https://github.com/abandonware/node-bluetooth-hci-socket/pull/5 :-)

rzr, 2019-09-17:
And released To npm

maxbellec, 2019-11-14:
```bash
npm install @abandonware/noble
``` 
works for me

rzr, 2019-11-14:
Please also review:
https://github.com/noble/node-bluetooth-hci-socket/pull/105
Then I will try to upstream more abandonware patches once this one is merged

Please review open PR to:
https://github.com/abandonware

rzr, 2021-01-06:
Feel free to comment this related ticket:

https://github.com/abandonware/node-bluetooth-hci-socket/issues/9

rzr, 2021-10-01:
May you forward your ticket to downstream version if relevant or confirm @abandonware version match your requirements here:

https://github.com/noble/node-bluetooth-hci-socket/issues/123

Liu-Jinshuai, 2023-09-12:
```
npm i bleno2
```
You can try this. It works well on my Linux. It has not been tested on other platforms. I hope it will help you.