noble/node-bluetooth-hci-socket #85: Publish on npm

DomiR, 2018-05-07:
Will you publish the current master any time soon?

morganrallen, 2018-05-30:
Yeah @sandeepmistry it would be great to see 0.5.2 published. I cannot get an earlier version to work on CHIP.

rzr, 2018-11-08:
Yes an intermediary version before node10 support will not hurt

rzr, 2018-12-17:
Last call before I try to upload a revision on npm org subrepo.

Please test:
https://github.com/abandonware/node-bluetooth-hci-socket/

I am listening all recommendations from upstream and others.

Apollon77, 2018-12-17:
Go from my side. I only had a very short look onto it ...

rzr, 2018-12-18:
@Apollon77  for CI check:
https://ci.appveyor.com/project/rzr/node-bluetooth-hci-socket

Apollon77, 2018-12-18:
cool, so you should remove nodejs 4 from testing :-)

rzr, 2018-12-18:
PR welcome ;)

rzr, 2018-12-18:
Please review:  
https://github.com/abandonware/noble/pulls