noble/node-bluetooth-hci-socket #88: Add to whitelist

MegaMech, 2018-05-17:
Hi,
I have an ASUS flip 14 (4gb pentium 4 core version)
It has a Qualcomm Atheros bluetooth adapter (4.1)
ID: 13D3, 3491.

the advertisement-discovery.js file worked with it. It was able to see my Kawai ES110 electric piano.

Is there any additional testing you would like me to do before you add it to the whitelist?

Thanks,

- Alex

jrobeson, 2018-06-18:
@MegaMech : you need to provide the underlying chip name like `Qualcomm Atheros QCA61x4` if possible as per https://github.com/noble/node-bluetooth-hci-socket/blob/master/lib/usb.js#L14

rzr, 2021-10-01:
May you forward your ticket to downstream version if relevant or confirm @abandonware version match your requirements here:

https://github.com/noble/node-bluetooth-hci-socket/issues/123