noble/node-bluetooth-hci-socket #93: Clang problem on termux

claudiobadii, 2018-07-02:
There seems to be a problem with installing the package on termux. Clang exception:

[image]

claudiobadii, 2018-07-03:
Hi,

I removed the clang package and I reinstalled gcc package from a third-part repository:

[https://github.com/its-pointless/gcc_termux](https://github.com/its-pointless/gcc_termux)

Now termux is blocked for "undefined reference to v8". Do you have any idea?
[image]

Happy0, 2018-08-09:
I'm also trying to get this working on android, @claudiobadii and have reached the same stage as you after switching form clang to gcc.

I'll let you know if I find a way to resolve this issue. I'm also tracking my progress here:

http://between-two-worlds.dk:8807/%250%2FPl9mdoYgP4FX%2FnyXxoDtLCDYBjCCEiN29vadRMFtc%3D.sha256

claudiobadii, 2018-08-10:
It's been a long time that I didn't develop in C/C++, but I had the luck to
help a student in these weeks just on these languages. The fact that it
doesn't find references to the various V8 libraries I think is due to the
fact that Termux lacks the V8 package and consequently its
libraries. That's why I ventured to create a package for termux, but even
there I found problems https://github.com/termux/termux-packages/issues/1750.
In the coming weeks I should try to implement Bluetooth on termux via
termux-api and JAVA.