noble/node-bluetooth-hci-socket #95: Raspberry pi zero W install issues

carlioth, 2018-08-03:
Hi. 

Im getting errors when trying to node-gyp install for arm6 (Raspberry pi zero w)

Could you point me in any direction of that to do, errors:

Node version= v8.9.3
npm version= 5.5.1

`../src/BluetoothHciSocket.cpp: At global scope:
../src/BluetoothHciSocket.cpp:365:11: error: expected constructor, destructor, or type conversion before '(' token
 NAN_METHOD(BluetoothHciSocket::New) {
           ^
../src/BluetoothHciSocket.cpp:374:11: error: expected constructor, destructor, or type conversion before '(' token
 NAN_METHOD(BluetoothHciSocket::Start) {
           ^
../src/BluetoothHciSocket.cpp:384:11: error: expected constructor, destructor, or type conversion before '(' token
 NAN_METHOD(BluetoothHciSocket::BindRaw) {
           ^
binding.target.mk:93: recipe for target 'Release/obj.target/binding/src/BluetoothHciSocket.o' failed
make: *** [Release/obj.target/binding/src/BluetoothHciSocket.o] Error 1
make: Leaving directory '/home/pi/git/pubnub-ruuvi/node_modules/bluetooth-hci-socket/build'
gyp ERR! build error
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/local/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:258:23)
gyp ERR! stack     at emitTwo (events.js:126:13)
gyp ERR! stack     at ChildProcess.emit (events.js:214:7)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:198:12)
gyp ERR! System Linux 4.14.52+
gyp ERR! command "/usr/local/bin/node" "/usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /home/pi/git/pubnub-ruuvi/node_modules/bluetooth-hci-socket
gyp ERR! node -v v8.9.3
gyp ERR! node-gyp -v v3.6.2
gyp ERR! not ok`

jptrsn, 2018-09-03:
I'm also unable to build on a Pi Zero W or an rPi 3B. Same error.

liquidcowgithub, 2018-09-22:
I had this issue as well when running on node 10.x on a Pi Zero W with Rasbian

Downgrading to node v8.11.1 seemed to do the trick.

Installed following the official docs
https://nodejs.org/en/download/package-manager/#debian-and-ubuntu-based-linux-distributions

st3xupery, 2018-11-14:
> Downgrading to node v8.11.1 seemed to do the trick.

Yes, this worked for me also.

jshaw, 2018-11-14:
I had to install `sudo apt-get install bluetooth bluez libbluetooth-dev libudev-dev`

rashidtn, 2018-12-10:
> I had this issue as well when running on node 10.x on a Pi Zero W with Rasbian
> 
> Downgrading to node v8.11.1 seemed to do the trick.
> 
> Installed following the official docs
> https://nodejs.org/en/download/package-manager/#debian-and-ubuntu-based-linux-distributions

same here

carlioth, 2020-09-08:
Closing due to age

closed 2020-09-08 by carlioth