noble/node-bluetooth-hci-socket #103: Error Installing Mojave 10.14.1

indigoswann, 2018-11-21:
XCode installed, all BLE core libraries available.  Keep getting

`make: *** [Release/obj.target/binding/src/XpcConnection.o] Error 1
gyp ERR! build error
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/local/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:262:23)
gyp ERR! stack     at ChildProcess.emit (events.js:182:13)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:240:12)
gyp ERR! System Darwin 18.2.0
gyp ERR! command "/usr/local/Cellar/node/10.11.0/bin/node" "/usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /Users/greggushard/Apps/vss-desktop/node_modules/xpc-connection
gyp ERR! node -v v10.11.0
gyp ERR! node-gyp -v v3.8.0
gyp ERR! not ok
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: bluetooth-hci-socket@0.5.1 (node_modules/bluetooth-hci-socket):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for bluetooth-hci-socket@0.5.1: wanted {"os":"linux,android,win32","arch":"any"} (current: {"os":"darwin","arch":"x64"})
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: xpc-connection@0.1.4 (node_modules/xpc-connection):
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: xpc-connection@0.1.4 install: `node-gyp rebuild`
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: Exit status 1`
`

goapilz, 2019-02-01:
I have the same error under ubuntu with node 11.

First i installed the bluetooth drivers via:
sudo apt-get install bluetooth bluez libbluetooth-dev libudev-dev

And then i installed noble with the bluetooth-hci-socket dependency
npm install noble

Pre-built binaries not found for usb@1.5.0 and node@11.6.0 (node-v67 ABI, glibc) (falling back to sourc                                                                                                                              e compile with node-gyp)
...
binding.target.mk:101: recipe for target 'Release/obj.target/binding/src/BluetoothHciSocket.o' failed
make: *** [Release/obj.target/binding/src/BluetoothHciSocket.o] Error 1
make: Leaving directory '/home/nodeUser/dev/node/bluetooth-test/node_modules/bluetooth-hci-socket/build'
gyp ERR! build error
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/home/nodeUser/.nvm/versions/node/v11.6.0/lib/node_modules/npm/node_modules/node                                                                                                                              -gyp/lib/build.js:262:23)
gyp ERR! stack     at ChildProcess.emit (events.js:188:13)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:254:12)
gyp ERR! System Linux 4.15.0-42-generic
gyp ERR! command "/home/nodeUser/.nvm/versions/node/v11.6.0/bin/node" "/home/nodeUser/.nvm/versions/node/v11.6.0/lib/node_mod                                                                                                                              ules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /home/nodeUser/dev/node/bluetooth-test/node_modules/bluetooth-hci-socket
gyp ERR! node -v v11.6.0
gyp ERR! node-gyp -v v3.8.0
gyp ERR! not ok

anthonysiano, 2019-02-19:
Same here, 

macOS 10.13.6

make: *** [Release/obj.target/binding/src/XpcConnection.o] Error 1
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/local/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:262:23)
gyp ERR! stack     at ChildProcess.emit (events.js:197:13)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:254:12)
gyp ERR! System Darwin 17.7.0
gyp ERR! command "/usr/local/Cellar/node/11.10.0/bin/node" "/usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /usr/local/lib/node_modules/noble/node_modules/xpc-connection
gyp ERR! node -v v11.10.0
gyp ERR! node-gyp -v v3.8.0
gyp ERR! not ok 
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: bluetooth-hci-socket@0.5.1 (node_modules/noble/node_modules/bluetooth-hci-socket):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for bluetooth-hci-socket@0.5.1: wanted {"os":"linux,android,win32","arch":"any"} (current: {"os":"darwin","arch":"x64"})
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: xpc-connection@0.1.4 (node_modules/noble/node_modules/xpc-connection):
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: xpc-connection@0.1.4 install: `node-gyp rebuild`
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: Exit status 1

janhaa, 2019-09-18:
Try this maintained package: https://github.com/abandonware/node-bluetooth-hci-socket