noble/node-bluetooth-hci-socket #107: doesn't compile on RPi 3B+

janeksz, 2019-01-31:
root@janek:~/.node-red# service nodered stop
root@janek:~/.node-red# uname -a
Linux janek 4.14.90-v7+ #1183 SMP Fri Dec 21 14:03:50 GMT 2018 armv7l GNU/Linux
root@janek:~/.node-red# npm install --unsafe-permit node-red-contrib-noble

> usb@1.5.0 install /root/.node-red/node_modules/usb
> node-pre-gyp install --fallback-to-build

node-pre-gyp WARN Using needle for node-pre-gyp https download
node-pre-gyp WARN Tried to download(404): https://github.com/tessel/node-usb/releases/download/1.5.0/usb_bindings-v1.5.0-node-v64-linux-arm.tar.gz
node-pre-gyp WARN Pre-built binaries not found for usb@1.5.0 and node@10.15.0 (node-v64 ABI, glibc) (falling back to source compile with node-gyp)
make: Entering directory '/root/.node-red/node_modules/usb/build'
  CC(target) Release/obj.target/libusb/libusb/libusb/core.o
  CC(target) Release/obj.target/libusb/libusb/libusb/descriptor.o
  CC(target) Release/obj.target/libusb/libusb/libusb/hotplug.o
  CC(target) Release/obj.target/libusb/libusb/libusb/io.o
  CC(target) Release/obj.target/libusb/libusb/libusb/strerror.o
  CC(target) Release/obj.target/libusb/libusb/libusb/sync.o
  CC(target) Release/obj.target/libusb/libusb/libusb/os/poll_posix.o
  CC(target) Release/obj.target/libusb/libusb/libusb/os/threads_posix.o
  CC(target) Release/obj.target/libusb/libusb/libusb/os/linux_usbfs.o
  CC(target) Release/obj.target/libusb/libusb/libusb/os/linux_udev.o
  AR(target) Release/obj.target/usb.a
  COPY Release/usb.a
  CXX(target) Release/obj.target/usb_bindings/src/node_usb.o
../src/node_usb.cc: In function ‘Nan::NAN_METHOD_RETURN_TYPE SetDebugLevel(Nan::NAN_METHOD_ARGS_TYPE)’:
../src/node_usb.cc:99:73: warning: ‘uint32_t v8::Value::Uint32Value() const’ is deprecated: Use maybe version [-Wdeprecated-declarations]
 if (info.Length() != 1 || !info[0]->IsUint32() || info[0]->Uint32Value() > 4) {
                                                                        ^
In file included from /root/.node-gyp/10.15.0/include/node/v8.h:26:0,
                 from ../src/node_usb.h:12,
                 from ../src/node_usb.cc:1:
/root/.node-gyp/10.15.0/include/node/v8.h:2477:47: note: declared here
   V8_DEPRECATED("Use maybe version", uint32_t Uint32Value() const);
                                               ^
/root/.node-gyp/10.15.0/include/node/v8config.h:324:3: note: in definition of macro ‘V8_DEPRECATED’
   declarator __attribute__((deprecated(message)))
   ^~~~~~~~~~
../src/node_usb.cc:103:53: warning: ‘uint32_t v8::Value::Uint32Value() const’ is deprecated: Use maybe version [-Wdeprecated-declarations]
  libusb_set_debug(usb_context, info[0]->Uint32Value());
                                                     ^
In file included from /root/.node-gyp/10.15.0/include/node/v8.h:26:0,
                 from ../src/node_usb.h:12,
                 from ../src/node_usb.cc:1:
/root/.node-gyp/10.15.0/include/node/v8.h:2477:47: note: declared here
   V8_DEPRECATED("Use maybe version", uint32_t Uint32Value() const);
                                               ^
/root/.node-gyp/10.15.0/include/node/v8config.h:324:3: note: in definition of macro ‘V8_DEPRECATED’
   declarator __attribute__((deprecated(message)))
   ^~~~~~~~~~
../src/node_usb.cc: In function ‘void handleHotplug(std::pair<libusb_device*, libusb_hotplug_event>)’:
../src/node_usb.cc:151:58: warning: ‘v8::Local<v8::Value> Nan::MakeCallback(v8::Local<v8::Object>, const char*, int, v8::Local<v8::Value>*)’ is deprecated [-Wdeprecated-declarations]
  Nan::MakeCallback(Nan::New(hotplugThis), "emit", 2, argv);
                                                          ^
In file included from ../src/helpers.h:3:0,
                 from ../src/node_usb.h:21,
                 from ../src/node_usb.cc:1:
../../nan/nan.h:1001:46: note: declared here
   NAN_DEPRECATED inline v8::Local<v8::Value> MakeCallback(
                                              ^~~~~~~~~~~~
../src/node_usb.cc: In function ‘v8::Local<v8::Value> libusbException(int)’:
../src/node_usb.cc:301:14: warning: ‘v8::Local<v8::Object> v8::Value::ToObject() const’ is deprecated: Use maybe version [-Wdeprecated-declarations]
  e->ToObject()->Set(Nan::New<String>("errno").ToLocalChecked(), Nan::New<Integer>(errorno));
              ^
In file included from ../src/node_usb.h:12:0,
                 from ../src/node_usb.cc:1:
/root/.node-gyp/10.15.0/include/node/v8.h:10046:15: note: declared here
 Local<Object> Value::ToObject() const {
               ^~~~~
  CXX(target) Release/obj.target/usb_bindings/src/device.o
[... 101 lines cut ...]
make: Leaving directory '/root/.node-red/node_modules/bluetooth-hci-socket/build'
gyp ERR! build error
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:262:23)
gyp ERR! stack     at ChildProcess.emit (events.js:182:13)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:240:12)
gyp ERR! System Linux 4.14.90-v7+
gyp ERR! command "/usr/bin/node" "/usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /root/.node-red/node_modules/bluetooth-hci-socket
gyp ERR! node -v v10.15.0
gyp ERR! node-gyp -v v3.8.0
gyp ERR! not ok
npm WARN node-red-project@0.0.1 No repository field.
npm WARN node-red-project@0.0.1 No license field.
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: xpc-connection@0.1.4 (node_modules/xpc-connection):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for xpc-connection@0.1.4: wanted {"os":"darwin","arch":"any"} (current: {"os":"linux","arch":"arm"})
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: bluetooth-hci-socket@0.5.1 (node_modules/bluetooth-hci-socket):
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: bluetooth-hci-socket@0.5.1 install: `node-gyp rebuild`
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: Exit status 1

+ node-red-contrib-noble@0.3.0
updated 1 package and audited 6537 packages in 59.264s
found 28 vulnerabilities (14 low, 9 moderate, 5 high)
  run `npm audit fix` to fix them, or `npm audit` for details
root@janek:~/.node-red#

goapilz, 2019-02-01:
I have the same error under ubuntu (AMD E350 CPU) with node 11.

First i installed the bluetooth drivers via:
sudo apt-get install bluetooth bluez libbluetooth-dev libudev-dev

And then i installed noble with the bluetooth-hci-socket dependency
npm install noble

Pre-built binaries not found for usb@1.5.0 and node@11.6.0 (node-v67 ABI, glibc) (falling back to sourc                                                                                                                              e compile with node-gyp)
...
binding.target.mk:101: recipe for target 'Release/obj.target/binding/src/BluetoothHciSocket.o' failed
make: *** [Release/obj.target/binding/src/BluetoothHciSocket.o] Error 1
make: Leaving directory '/home/nodeUser/dev/node/bluetooth-test/node_modules/bluetooth-hci-socket/build'
gyp ERR! build error
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/home/nodeUser/.nvm/versions/node/v11.6.0/lib/node_modules/npm/node_modules/node                                                                                                                              -gyp/lib/build.js:262:23)
gyp ERR! stack     at ChildProcess.emit (events.js:188:13)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:254:12)
gyp ERR! System Linux 4.15.0-42-generic
gyp ERR! command "/home/nodeUser/.nvm/versions/node/v11.6.0/bin/node" "/home/nodeUser/.nvm/versions/node/v11.6.0/lib/node_mod                                                                                                                              ules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /home/nodeUser/dev/node/bluetooth-test/node_modules/bluetooth-hci-socket
gyp ERR! node -v v11.6.0
gyp ERR! node-gyp -v v3.8.0
gyp ERR! not ok

quito96, 2019-02-06:
i have the same problem on the same device : 
pi@quitos_rpi3:~ $ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)"
NAME="Raspbian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"
ID=raspbian
ID_LIKE=debian

janeksz, 2019-02-06:
the temporary solution is to downgrade nodejs to version 9, it works for me. I was told it is a known problem and will be fixed (???)

Webunity, 2019-02-20:
See here for a possible fix:
https://github.com/noble/node-bluetooth-hci-socket/issues/84
https://github.com/noble/node-bluetooth-hci-socket/pull/91

atc1441, 2019-04-11:
Hi does anybody have found another solution for this? i still dont get it to work even with the downgrade to node 9.

And everything BLE relatet is build on the hci socket.

janhaa, 2019-09-18:
Fixed here: https://github.com/abandonware/node-bluetooth-hci-socket/pull/5

dbryar, 2019-12-11:
Confirmed working with abandonware patch

To use in noble, navigate to your project folder and
```
nano node_modules/noble/lib/hci-socket/hci.js
```
edit line 6
```
var BluetoothHciSocket = require('**@abandonware/**bluetooth-hci-socket');
```
and then install the abandonware update to get your project working (again)
```
npm install @abandonware/bluetooth-hci-socket
```

morkeltry, 2019-12-16:
Or, aliasing the package name instead of changing `require`s in the code:

`npm install bluetooth-hci-socket@npm:@abandonware/bluetooth-hci-socket`