noble/node-bluetooth-hci-socket #110: was compiled against different Node.js version

Karmac, 2019-04-12:
Hi!

I've been trying to install _bleno_  without success. I'm on a Raspberry PI 3 Model B. First I downgraded Node.js to version 8.15.1 in order to avoid node-gyp compiling error. Now, I get the following when running the program:

> App threw an error during load
Error: The module '/home/pi/Downloads/node_modules/@abandonware/bluetooth-hci-socket/lib/binding/binding.node'
was compiled against a different Node.js version using
NODE_MODULE_VERSION 57. This version of Node.js requires
NODE_MODULE_VERSION 64. Please try re-compiling or re-installing
the module (for instance, using `npm rebuild` or `npm install`).
    at process.module.(anonymous function) [as dlopen] (ELECTRON_ASAR.js:166:20)
    at Object.Module._extensions..node (internal/modules/cjs/loader.js:740:18)
    at Object.module.(anonymous function) [as .node] (ELECTRON_ASAR.js:166:20)
    at Module.load (internal/modules/cjs/loader.js:620:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:559:12)
    at Function.Module._load (internal/modules/cjs/loader.js:551:3)
    at Module.require (internal/modules/cjs/loader.js:658:17)
    at require (internal/modules/cjs/helpers.js:20:18)
    at Object.<anonymous> (/home/pi/Downloads/node_modules/@abandonware/bluetooth-hci-socket/lib/native.js:6:15)
    at Object.<anonymous> (/home/pi/Downloads/node_modules/@abandonware/bluetooth-hci-socket/lib/native.js:21:3)
A JavaScript error occurred in the main process
Uncaught Exception:
Error: The module '/home/pi/Downloads/node_modules/@abandonware/bluetooth-hci-socket/lib/binding/binding.node'
was compiled against a different Node.js version using
NODE_MODULE_VERSION 57. This version of Node.js requires
NODE_MODULE_VERSION 64. Please try re-compiling or re-installing
the module (for instance, using `npm rebuild` or `npm install`).
    at process.module.(anonymous function) [as dlopen] (ELECTRON_ASAR.js:166:20)
    at Object.Module._extensions..node (internal/modules/cjs/loader.js:740:18)
    at Object.module.(anonymous function) [as .node] (ELECTRON_ASAR.js:166:20)
    at Module.load (internal/modules/cjs/loader.js:620:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:559:12)
    at Function.Module._load (internal/modules/cjs/loader.js:551:3)
    at Module.require (internal/modules/cjs/loader.js:658:17)
    at require (internal/modules/cjs/helpers.js:20:18)
    at Object.<anonymous> (/home/pi/Downloads/node_modules/@abandonware/bluetooth-hci-socket/lib/native.js:6:15)
    at Object.<anonymous> (/home/pi/Downloads/node_modules/@abandonware/bluetooth-hci-socket/lib/native.js:21:3)`

In this one I was using @abandonware's fork. I've also switched versions, and I get the same error (compiled using XX and required version 64). I've already searched on the internet and I don't find any working solution.

If someone could please help me or give some ideas I would appreciate. Everything should work fine here because I've seen bleno installed on Raspberries.

Thanks,

psambit9791, 2019-05-03:
Have seen the same error. On diagnosing, figured that the problem was with using nvm for version management. NVM keeps the node versions in the directory: /usr/<username>/.nvm which is used for compiling when installing the packages.
Using the bluetooth-hci-socket requires using sudo command, which does not access the Node from nvm path, in stead uses the node version installed for root user.

For me, it appeared like this:
$ sudo node -v
v10.15.3
$ node -v
v9.11.2

**Solution**

Downgrade the node version for root to Node 8.15.1.

Karmac, 2019-05-15:
Hi @psambit9791, thanks for your response!

There was a mismatch on both versions. I downgraded NVM version because it was higher. Now I get the following:
node -v
v8.11.1
sudo node -v
v8.11.1

But I still get "Compiled against NODE_MODULE_VERSION 57. This version requires NODE_MODULE_VERSION 64".

I think I will try to uninstall NVM and use only system node.js. I'll let you know if I find something.

Karmac, 2019-06-12:
Okey, I finally solved the issue. Seems that Node.js version didn't even matter, because Electron uses it's own. I downgraded Electron to 2.0.18 which uses Node 8.9.x.

Thanks for your help.

closed 2019-06-12 by Karmac

komorebi-san, 2019-07-12:
@Karmac 

Can you elaborate more on how you recompile?