noble/node-bluetooth-hci-socket #112: npm-gyp rebuild fails - Debian

Waffaru, 2019-06-19:
Hello,

I'm trying to install bleno-package [(link)](https://github.com/noble/bleno) when it gets to the part where it tries installing this dependency, the build script fails.

Here's the error trace:
```

../src/BluetoothHciSocket.cpp: In static member function ‘static Nan::NAN_METHOD_RETURN_TYPE BluetoothHciSocket::BindRaw(Nan::NAN_METHOD_ARGS_TYPE)’:
../src/BluetoothHciSocket.cpp:395:34: error: no matching function for call to ‘v8::Value::IntegerValue()’
       devId = arg0->IntegerValue();
                                  ^
In file included from /home/pi/.node-gyp/12.4.0/include/node/node.h:63:0,
                 from /home/pi/.node-gyp/12.4.0/include/node/node_buffer.h:25,
                 from ../src/BluetoothHciSocket.cpp:7:
/home/pi/.node-gyp/12.4.0/include/node/v8.h:2565:40: note: candidate: v8::Maybe<long long int> v8::Value::IntegerValue(v8::Local<v8::Context>) const
   V8_WARN_UNUSED_RESULT Maybe<int64_t> IntegerValue(
                                        ^~~~~~~~~~~~
/home/pi/.node-gyp/12.4.0/include/node/v8.h:2565:40: note:   candidate expects 1 argument, 0 provided
../src/BluetoothHciSocket.cpp: In static member function ‘static Nan::NAN_METHOD_RETURN_TYPE BluetoothHciSocket::BindUser(Nan::NAN_METHOD_ARGS_TYPE)’:
../src/BluetoothHciSocket.cpp:417:34: error: no matching function for call to ‘v8::Value::IntegerValue()’
       devId = arg0->IntegerValue();
                                  ^
In file included from /home/pi/.node-gyp/12.4.0/include/node/node.h:63:0,
                 from /home/pi/.node-gyp/12.4.0/include/node/node_buffer.h:25,
                 from ../src/BluetoothHciSocket.cpp:7:
/home/pi/.node-gyp/12.4.0/include/node/v8.h:2565:40: note: candidate: v8::Maybe<long long int> v8::Value::IntegerValue(v8::Local<v8::Context>) const
   V8_WARN_UNUSED_RESULT Maybe<int64_t> IntegerValue(
                                        ^~~~~~~~~~~~
/home/pi/.node-gyp/12.4.0/include/node/v8.h:2565:40: note:   candidate expects 1 argument, 0 provided
binding.target.mk:105: recipe for target 'Release/obj.target/binding/src/BluetoothHciSocket.o' failed
make: *** [Release/obj.target/binding/src/BluetoothHciSocket.o] Error 1
make: Leaving directory '/home/pi/blezr/bleno/node_modules/bluetooth-hci-socket/build'
gyp ERR! build error
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/home/pi/.nvm/versions/node/v12.4.0/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:262:23)
gyp ERR! stack     at ChildProcess.emit (events.js:200:13)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:272:12)
gyp ERR! System Linux 4.19.42-v7+
gyp ERR! command "/home/pi/.nvm/versions/node/v12.4.0/bin/node" "/home/pi/.nvm/versions/node/v12.4.0/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /home/pi/blezr/bleno/node_modules/bluetooth-hci-socket
gyp ERR! node -v v12.4.0
gyp ERR! node-gyp -v v3.8.0
gyp ERR! not ok
npm WARN blezr-bleno@1.0.0 No repository field.
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: xpc-connection@0.1.4 (node_modules/xpc-connection):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for xpc-connection@0.1.4: wanted {"os":"darwin","arch":"any"} (current: {"os":"linux","arch":"arm"})

npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! bluetooth-hci-socket@0.5.1 install: `node-gyp rebuild`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the bluetooth-hci-socket@0.5.1 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2019-06-19T09_19_19_689Z-debug.log
```

Any suggestions? I reinstalled npm-gyp and node prior to this. They are fresh installs.

Waffaru, 2019-06-20:
The original attempt was with node 12.4.0 . I tried this first with node 10 and had also issues so I finally used node version 8.16.0 (by changing my current version to that with nvm) and managed to get it working.

Leaving this open awaiting for a response to this, but if you're having issues going back to an earlier version of node seems to work.

johnwalicki, 2019-07-03:
Compiling bluetooth-hci-socket on node v10.16 throws this error
```
../src/BluetoothHciSocket.cpp: In member function ‘void BluetoothHciSocket::emitErrnoError()’:
../src/BluetoothHciSocket.cpp:275:72: error: no matching function for call to ‘v8::Function::NewInstance(int, v8::Local<v8::Value> [1])’
   Local<Value> error = errorConstructor->NewInstance(1, constructorArgs);
                                                                        ^
/home/pi/.node-gyp/10.16.0/include/node/v8.h:3996:44: note:   candidate expects 1 argument, 2 provided

```

johnwalicki, 2019-07-03:
Starting with node 10, v8.h NewInstance() requires an additional argument 
 Local<Context> context,

johnwalicki, 2019-07-03:
I don't know enough about BluetoothHciSocket.cpp to fix it however.

johnwalicki, 2019-07-03:
My local "fix" comments out the contents of the BluetoothHciSocket::emitErrnoError() so that the native module compiles.  That's not a recommended technique.

johnnyxwan, 2019-07-30:
> The original attempt was with node 12.4.0 . I tried this first with node 10 and had also issues so I finally used node version 8.16.0 (by changing my current version to that with nvm) and managed to [...]

Not sure if it is my issue only, I failed to compile even with 8.16.0

`> bluetooth-hci-socket@0.5.1 install /home/pi/testv8/node_modules/bluetooth-hci-socket
> node-gyp rebuild

make: Entering directory '/home/pi/testv8/node_modules/bluetooth-hci-socket/build'
  CXX(target) Release/obj.target/binding/src/BluetoothHciSocket.o
In file included from /home/pi/.node-gyp/8.16.0/include/node/node.h:63,
                 from /home/pi/.node-gyp/8.16.0/include/node/node_buffer.h:25,
                 from ../src/BluetoothHciSocket.cpp:7:
/home/pi/.node-gyp/8.16.0/include/node/v8.h: In static member function ‘static void v8::V8::SetAllowCodeGenerationFromStringsCallback(v8::DeprecatedAllowCodeGenerationFromStringsCallback)’:
/home/pi/.node-gyp/8.16.0/include/node/v8.h:10304:19: warning: cast between incompatible function types from ‘v8::DeprecatedAllowCodeGenerationFromStringsCallback’ {aka ‘bool (*)(v8::Local<v8::Context>)’} to ‘v8::FreshNewAllo     wCodeGenerationFromStringsCallback’ {aka ‘bool (*)(v8::Local<v8::Context>, v8::Local<v8::String>)’} [-Wcast-function-type]
           callback));
                   ^
/home/pi/.node-gyp/8.16.0/include/node/v8.h: In static member function ‘static void v8::V8::RemoveGCPrologueCallback(v8::GCCallback)’:
/home/pi/.node-gyp/8.16.0/include/node/v8.h:10349:53: warning: cast between incompatible function types from ‘v8::GCCallback’ {aka ‘void (*)(v8::GCType, v8::GCCallbackFlags)’} to ‘v8::Isolate::GCCallback’ {aka ‘void (*)(v8::I     solate*, v8::GCType, v8::GCCallbackFlags)’} [-Wcast-function-type]
       reinterpret_cast<Isolate::GCCallback>(callback));
                                                     ^
/home/pi/.node-gyp/8.16.0/include/node/v8.h: In static member function ‘static void v8::V8::RemoveGCEpilogueCallback(v8::GCCallback)’:
/home/pi/.node-gyp/8.16.0/include/node/v8.h:10356:53: warning: cast between incompatible function types from ‘v8::GCCallback’ {aka ‘void (*)(v8::GCType, v8::GCCallbackFlags)’} to ‘v8::Isolate::GCCallback’ {aka ‘void (*)(v8::I     solate*, v8::GCType, v8::GCCallbackFlags)’} [-Wcast-function-type]
       reinterpret_cast<Isolate::GCCallback>(callback));
                                                     ^
In file included from ../src/BluetoothHciSocket.cpp:8:
../../nan/nan.h: In function ‘void Nan::AsyncQueueWorker(Nan::AsyncWorker*)’:
../../nan/nan.h:2298:62: warning: cast between incompatible function types from ‘void (*)(uv_work_t*)’ {aka ‘void (*)(uv_work_s*)’} to ‘uv_after_work_cb’ {aka ‘void (*)(uv_work_s*, int)’} [-Wcast-function-type]
     , reinterpret_cast<uv_after_work_cb>(AsyncExecuteComplete)
                                                              ^
../src/BluetoothHciSocket.cpp: In member function ‘void BluetoothHciSocket::poll()’:
../src/BluetoothHciSocket.cpp:251:95: warning: ‘v8::Local<v8::Value> Nan::MakeCallback(v8::Local<v8::Object>, v8::Local<v8::String>, int, v8::Local<v8::Value>*)’ is deprecated [-Wdeprecated-declarations]
     Nan::MakeCallback(Nan::New<Object>(this->This), Nan::New("emit").ToLocalChecked(), 2, argv);
                                                                                               ^
In file included from ../src/BluetoothHciSocket.cpp:8:
../../nan/nan.h:1045:46: note: declared here
   NAN_DEPRECATED inline v8::Local<v8::Value> MakeCallback(
                                              ^~~~~~~~~~~~
../src/BluetoothHciSocket.cpp:251:95: warning: ‘v8::Local<v8::Value> Nan::MakeCallback(v8::Local<v8::Object>, v8::Local<v8::String>, int, v8::Local<v8::Value>*)’ is deprecated [-Wdeprecated-declarations]
     Nan::MakeCallback(Nan::New<Object>(this->This), Nan::New("emit").ToLocalChecked(), 2, argv);
                                                                                               ^
In file included from ../src/BluetoothHciSocket.cpp:8:
../../nan/nan.h:1045:46: note: declared here
   NAN_DEPRECATED inline v8::Local<v8::Value> MakeCallback(
                                              ^~~~~~~~~~~~
../src/BluetoothHciSocket.cpp: In member function ‘void BluetoothHciSocket::emitErrnoError()’:
../src/BluetoothHciSocket.cpp:275:72: warning: ‘v8::Local<v8::Object> v8::Function::NewInstance(int, v8::Local<v8::Value>*) const’ is deprecated: Use maybe version [-Wdeprecated-declarations]
   Local<Value> error = errorConstructor->NewInstance(1, constructorArgs);
                                                                        ^
In file included from /home/pi/.node-gyp/8.16.0/include/node/v8.h:26,
                 from /home/pi/.node-gyp/8.16.0/include/node/node.h:63,
                 from /home/pi/.node-gyp/8.16.0/include/node/node_buffer.h:25,
                 from ../src/BluetoothHciSocket.cpp:7:
/home/pi/.node-gyp/8.16.0/include/node/v8.h:3897:31: note: declared here
                 Local<Object> NewInstance(int argc, Local<Value> argv[]) const);
                               ^~~~~~~~~~~
/home/pi/.node-gyp/8.16.0/include/node/v8config.h:318:3: note: in definition of macro ‘V8_DEPRECATED’
   declarator __attribute__((deprecated(message)))
   ^~~~~~~~~~
../src/BluetoothHciSocket.cpp:282:93: warning: ‘v8::Local<v8::Value> Nan::MakeCallback(v8::Local<v8::Object>, v8::Local<v8::String>, int, v8::Local<v8::Value>*)’ is deprecated [-Wdeprecated-declarations]
[... cut ...]
/home/pi/.node-gyp/8.16.0/include/node/v8.h: In instantiation of ‘void v8::PersistentBase<T>::SetWeak(P*, typename v8::WeakCallbackInfo<P>::Callback, v8::WeakCallbackType) [with P = Nan::ObjectWrap; T = v8::Object; typename v     8::WeakCallbackInfo<P>::Callback = void (*)(const v8::WeakCallbackInfo<Nan::ObjectWrap>&)]’:
../../nan/nan_object_wrap.h:65:61:   required from here
/home/pi/.node-gyp/8.16.0/include/node/v8.h:9313:16: warning: cast between incompatible function types from ‘v8::WeakCallbackInfo<Nan::ObjectWrap>::Callback’ {aka ‘void (*)(const v8::WeakCallbackInfo<Nan::ObjectWrap>&)’} to       Callback’ {aka ‘void (*)(const v8::WeakCallbackInfo<void>&)’} [-Wcast-function-type]
`

johnnyxwan, 2019-07-30:
Finally figure it out, I forgot to execute
`npm install -g node-gyp`
after nvm use

playground, 2019-08-05:
I'm having the same issue.  Am running node v11.15.0 on MacOS MoJave 10.14.6
I've tried installing Xcode and node-gyp again but didn't help.

Waffaru, 2019-08-06:
> 
> 
> I'm having a similar issue. Am running node v11.15.0 on MacOS MoJave 10.14.6
> I've tried installing Xcode and node-gyp again but didn't help.
> 
> ```
> > xpc-connection@0.1.4 install /Users/tester/git_repo/sandbox/wu/playbox/node-web-bluetooth-test/node_modules/xpc-connection
> > node-gyp rebuild
> 
>   CXX(target) Release/obj.target/binding/src/XpcConnection.o
> ../src/XpcConnection.cpp:103:41: warning: 'IntegerValue' is deprecated: Use maybe version [-Wdeprecated-declarations]
>     xpcObject = xpc_int64_create(value->IntegerValue());
>                                         ^
> /Users/tester/Library/Caches/node-gyp/11.15.0/include/node/v8.h:2570:3: note: 'IntegerValue' has been explicitly marked deprecated here
>   V8_DEPRECATED("Use maybe version", int64_t IntegerValue() const);
>   ^
> /Users/tester/Library/Caches/node-gyp/11.15.0/include/node/v8config.h:326:29: note: expanded from macro 'V8_DEPRECATED'
>   declarator __attribute__((deprecated(message)))
>                             ^
> ../src/XpcConnection.cpp:113:40: warning: 'ToObject' is deprecated: Use maybe version [-Wdeprecated-declarations]
>     Local<Object> valueObject = value->ToObject();
>                                        ^
> /Users/tester/Library/Caches/node-gyp/11.15.0/include/node/v8.h:2549:10: note: 'ToObject' has been explicitly marked deprecated here
>   inline V8_DEPRECATED("Use maybe version", Local<Object> ToObject() const);
>          ^
> /Users/tester/Library/Caches/node-gyp/11.15.0/include/node/v8config.h:326:29: note: expanded from macro 'V8_DEPRECATED'
>   declarator __attribute__((deprecated(message)))
>                             ^
> ../src/XpcConnection.cpp:123:40: warning: 'ToObject' is deprecated: Use maybe version [-Wdeprecated-declarations]
>     Local<Object> valueObject = value->ToObject();
>                                        ^
> /Users/tester/Library/Caches/node-gyp/11.15.0/include/node/v8.h:2549:10: note: 'ToObject' has been explicitly marked deprecated here
>   inline V8_DEPRECATED("Use maybe version", Local<Object> ToObject() const);
>          ^
> /Users/tester/Library/Caches/node-gyp/11.15.0/include/node/v8config.h:326:29: note: expanded from macro 'V8_DEPRECATED'
>   declarator __attribute__((deprecated(message)))
>                             ^
> ../src/XpcConnection.cpp:143:89: error: too few arguments to function call, expected 2, have 1
>       Local<Value> propertyValue = object->GetRealNamedProperty(propertyName->ToString());
>                                    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~                         ^
> /Users/tester/Library/Caches/node-gyp/11.15.0/include/node/v8.h:3668:3: note: 'GetRealNamedProperty' declared here
>   V8_WARN_UNUSED_RESULT MaybeLocal<Value> GetRealNamedProperty(
>   ^
> /Users/tester/Library/Caches/node-gyp/11.15.0/include/node/v8config.h:418:31: note: expanded from macro 'V8_WARN_UNUSED_RESULT'
> #define V8_WARN_UNUSED_RESULT __attribute__((warn_unused_result))
>                               ^
> ../src/XpcConnection.cpp:143:79: warning: 'ToString' is deprecated: Use maybe version [-Wdeprecated-declarations]
>       Local<Value> propertyValue = object->GetRealNamedProperty(propertyName->ToString());
>                                                                               ^
> /Users/tester/Library/Caches/node-gyp/11.15.0/include/node/v8.h:2548:10: note: 'ToString' has been explicitly marked deprecated here
>   inline V8_DEPRECATED("Use maybe version", Local<String> ToString() const);
>          ^
> /Users/tester/Library/Caches/node-gyp/11.15.0/include/node/v8config.h:326:29: note: expanded from macro 'V8_DEPRECATED'
>   declarator __attribute__((deprecated(message)))
>                             ^
> ../src/XpcConnection.cpp:254:12: warning: 'MakeCallback' is deprecated [-Wdeprecated-declarations]
>       Nan::MakeCallback(Nan::New<Object>(this->This), Nan::New("emit").ToLocalChecked(), 2, argv);
>            ^
> ../../nan/nan.h:1045:3: note: 'MakeCallback' has been explicitly marked deprecated here
>   NAN_DEPRECATED inline v8::Local<v8::Value> MakeCallback(
>   ^
> ../../nan/nan.h:104:40: note: expanded from macro 'NAN_DEPRECATED'
> # define NAN_DEPRECATED __attribute__((deprecated))
>                                        ^
> ../src/XpcConnection.cpp:263:12: warning: 'MakeCallback' is deprecated [-Wdeprecated-declarations]
>       Nan::MakeCallback(Nan::New<Object>(this->This), Nan::New("emit").ToLocalChecked(), 2, argv);
>            ^
> ../../nan/nan.h:1045:3: note: 'MakeCallback' has been explicitly marked deprecated here
>   NAN_DEPRECATED inline v8::Local<v8::Value> MakeCallback(
>   ^
[... 9 lines cut ...]
> gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:255:12)
> gyp ERR! System Darwin 18.7.0
> gyp ERR! command "/Users/tester/.nvm/versions/node/v11.15.0/bin/node" "/Users/tester/.nvm/versions/node/v11.15.0/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
> gyp ERR! cwd /Users/tester/git_repo/sandbox/wu/playbox/node-web-bluetooth-test/node_modules/xpc-connection
> gyp ERR! node -v v11.15.0
> gyp ERR! node-gyp -v v5.0.3
> gyp ERR! not ok 
> npm WARN saveError ENOENT: no such file or directory, open '/Users/tester/git_repo/sandbox/wu/playbox/node-web-bluetooth-test/package.json'
> npm WARN enoent ENOENT: no such file or directory, open '/Users/tester/git_repo/sandbox/wu/playbox/node-web-bluetooth-test/package.json'
> npm WARN node-web-bluetooth-test No description
[... cut ...]
> npm WARN optional SKIPPING OPTIONAL DEPENDENCY: xpc-connection@0.1.4 install: `node-gyp rebuild`
> npm WARN optional SKIPPING OPTIONAL DEPENDENCY: Exit status 1
> 
> + node-web-bluetooth@1.1.0
> added 8 packages from 7 contributors and audited 125 packages in 4.876s
> found 2 low severity vulnerabilities
> ```

Heya, if you don't absolutely need that version of node downgrading to node 8 and then rebuilding worked for me.  It seems to be an issue with HciBluetoothSocket not supporting newer versions of Node. Look at the answers by  johnwalicki above for more info.

playground, 2019-08-06:
I applied the workaround from pull request #26 by @taoyuan fixes the problem for now.  I'm curious if this will get addressed soon.