noble/node-bluetooth-hci-socket #123: Maintenance status ?

rzr, 2020-01-24:
Hi,

Is there any interest to keep maintaining this software ? 

If yes please review/merge or pick changes from:
https://github.com/noble/node-bluetooth-hci-socket/pull/122

If not a "community fork" is maintained else where:

https://abandonware.github.io/ 

Relate-to: abandonware/abandonware.github.io#5 Regards

rzr, 2020-01-24:
Our latest release is:

https://www.npmjs.com/package/@abandonware/bluetooth-hci-socket/v/0.5.3-4

rzr, 2020-02-26:
reviews welcome:

https://github.com/noble/node-bluetooth-hci-socket/pull/122

rzr, 2020-03-05:
FYI an updated version has been released:

https://www.npmjs.com/package/@abandonware/bluetooth-hci-socket/v/0.5.3-5

rzr, 2021-10-01:
update: New downstream release:

https://www.npmjs.com/package/@abandonware/bluetooth-hci-socket/v/0.5.3-8

Relate-to: https://github.com/abandonware/node-bluetooth-hci-socket/issues/30

ebaauw, 2021-10-01:
> Is there any interest to keep maintaining this software ?

Yes, please.  Currently using the downstream fork by abandonware.

rzr, 2024-04-12:
upstream feedback welcome,

meanwhile free to use our latest release:

https://www.npmjs.com/package/@abandonware/bluetooth-hci-socket/v/0.5.3-12