noble/node-bluetooth-hci-socket #128: raspberry pi3

glavermicocca, 2020-06-12:
it doesn't work -> sudo npm install bluetooth-hci-socket

binding.target.mk:109: set di istruzioni per l'obiettivo "Release/obj.target/binding/src/BluetoothHciSocket.o" non riuscito
make: *** [Release/obj.target/binding/src/BluetoothHciSocket.o] Errore 1
make: uscita dalla directory "/home/pi/node-bluetooth-hci-socket/build"
gyp ERR! build error
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:194:23)
gyp ERR! stack     at ChildProcess.emit (events.js:315:20)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:275:12)
gyp ERR! System Linux 4.19.66-v7+
gyp ERR! command "/usr/bin/node" "/usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "build" "--fallback-to-build" "--module=/home/pi/node-bluetooth-hci-socket/lib/binding/binding.node" "--module_name=binding" "--module_path=/home/pi/node-bluetooth-hci-socket/lib/binding"
gyp ERR! cwd /home/pi/node-bluetooth-hci-socket
gyp ERR! node -v v12.18.0
gyp ERR! node-gyp -v v5.1.0
gyp ERR! not ok
node-pre-gyp ERR! build error
node-pre-gyp ERR! stack Error: Failed to execute '/usr/bin/node /usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js build --fallback-to-build --module=/home/pi/node-bluetooth-hci-socket/lib/binding/binding.node --module_name=binding --module_path=/home/pi/node-bluetooth-hci-socket/lib/binding' (1)
node-pre-gyp ERR! stack     at ChildProcess.<anonymous> (/home/pi/node-bluetooth-hci-socket/node_modules/node-pre-gyp/lib/util/compile.js:83:29)
node-pre-gyp ERR! stack     at ChildProcess.emit (events.js:315:20)
node-pre-gyp ERR! stack     at maybeClose (internal/child_process.js:1021:16)
node-pre-gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:286:5)
node-pre-gyp ERR! System Linux 4.19.66-v7+
node-pre-gyp ERR! command "/usr/bin/node" "/home/pi/node-bluetooth-hci-socket/node_modules/.bin/node-pre-gyp" "install" "--fallback-to-build"
node-pre-gyp ERR! cwd /home/pi/node-bluetooth-hci-socket
node-pre-gyp ERR! node -v v12.18.0
node-pre-gyp ERR! node-pre-gyp -v v0.6.39
node-pre-gyp ERR! not ok
Failed to execute '/usr/bin/node /usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js build --fallback-to-build --module=/home/pi/node-bluetooth-hci-socket/lib/binding/binding.node --module_name=binding --module_path=/home/pi/node-bluetooth-hci-socket/lib/binding' (1)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! bluetooth-hci-socket@0.5.2 install: `node-pre-gyp install --fallback-to-build`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the bluetooth-hci-socket@0.5.2 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2020-06-12T15_37_12_946Z-debug.log
pi@raspiTermo2:~/node-bluetooth-hci-socket $ binding.target.mk:109: set di istruzioni per l'obiettivo "Release/obj.target/binding/src/BluetoothHciSocket.o" non riuscito
make: *** [Release/obj.target/binding/src/BluetoothHciSocket.o] Errore 1
make: uscita dalla directory "/home/pi/node-bluetooth-hci-socket/build"
gyp ERR! build error
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:194:23)
gyp ERR! stack     at ChildProcess.emit (events.js:315:20)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:275:12)
gyp ERR! System Linux 4.19.66-v7+
gyp ERR! command "/usr/bin/node" "/usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "build" "--fallback-to-build" "--module=/home/pi/node-bluetooth-hci-socket/lib/binding/binding.node" "--module_name=binding" "--module_path=/home/pi/node-bluetooth-hci-socket/lib/binding"
gyp ERR! cwd /home/pi/node-bluetooth-hci-socket
gyp ERR! node -v v12.18.0
gyp ERR! node-gyp -v v5.1.0
gyp ERR! not ok

SymbolCsaba, 2020-07-23:
Same issue

syntheticencounters, 2020-09-17:
I had to downgrade node to version 8.9.4 to get a successful install

rzr, 2021-10-01:
May you forward your ticket to downstream version if relevant or confirm @abandonware version match your requirements here:

https://github.com/noble/node-bluetooth-hci-socket/issues/123