noble/node-bluetooth-hci-socket #160: Installation fails (Arch Linux, node-gyp 9.3.0)

trougnouf, 2022-11-22:
npm installation fails on Arch Linux. I've tried with nodejs versions (19.1.0, 14.21, 16.18) and I have node-gyp 9.3.0 installed

<details><summary>Terminal output:</summary>

```
[trougnouf@d]: ~>$ npm install -g bluetooth-hci-socket
npm ERR! code 1
npm ERR! path /home/trougnouf/.local/lib/node_modules/bluetooth-hci-socket
npm ERR! command failed
npm ERR! command sh -c -- node-gyp rebuild
npm ERR! make: Entering directory '/home/trougnouf/.local/lib/node_modules/bluetooth-hci-socket/build'
npm ERR!   CXX(target) Release/obj.target/binding/src/BluetoothHciSocket.o
npm ERR! make: Leaving directory '/home/trougnouf/.local/lib/node_modules/bluetooth-hci-socket/build'
npm ERR! gyp info it worked if it ends with ok
npm ERR! gyp info using node-gyp@9.3.0
npm ERR! gyp info using node@19.1.0 | linux | x64
npm ERR! gyp info find Python using Python version 3.10.8 found at "/usr/bin/python3"
npm ERR! gyp info spawn /usr/bin/python3
npm ERR! gyp info spawn args [
npm ERR! gyp info spawn args   '/usr/lib/node_modules/node-gyp/gyp/gyp_main.py',
npm ERR! gyp info spawn args   'binding.gyp',
npm ERR! gyp info spawn args   '-f',
npm ERR! gyp info spawn args   'make',
npm ERR! gyp info spawn args   '-I',
npm ERR! gyp info spawn args   '/home/trougnouf/.local/lib/node_modules/bluetooth-hci-socket/build/config.gypi',
npm ERR! gyp info spawn args   '-I',
npm ERR! gyp info spawn args   '/usr/lib/node_modules/node-gyp/addon.gypi',
npm ERR! gyp info spawn args   '-I',
npm ERR! gyp info spawn args   '/home/trougnouf/.cache/node-gyp/19.1.0/include/node/common.gypi',
npm ERR! gyp info spawn args   '-Dlibrary=shared_library',
npm ERR! gyp info spawn args   '-Dvisibility=default',
npm ERR! gyp info spawn args   '-Dnode_root_dir=/home/trougnouf/.cache/node-gyp/19.1.0',
npm ERR! gyp info spawn args   '-Dnode_gyp_dir=/usr/lib/node_modules/node-gyp',
npm ERR! gyp info spawn args   '-Dnode_lib_file=/home/trougnouf/.cache/node-gyp/19.1.0/<(target_arch)/node.lib',
npm ERR! gyp info spawn args   '-Dmodule_root_dir=/home/trougnouf/.local/lib/node_modules/bluetooth-hci-socket',
npm ERR! gyp info spawn args   '-Dnode_engine=v8',
npm ERR! gyp info spawn args   '--depth=.',
npm ERR! gyp info spawn args   '--no-parallel',
npm ERR! gyp info spawn args   '--generator-output',
npm ERR! gyp info spawn args   'build',
npm ERR! gyp info spawn args   '-Goutput_dir=.'
npm ERR! gyp info spawn args ]
npm ERR! gyp info spawn make
npm ERR! gyp info spawn args [ 'BUILDTYPE=Release', '-C', 'build' ]
npm ERR! ../src/BluetoothHciSocket.cpp: In static member function ‘static void BluetoothHciSocket::Init(Nan::ADDON_REGISTER_FUNCTION_ARGS_TYPE)’:
npm ERR! ../src/BluetoothHciSocket.cpp:128:81: error: no matching function for call to ‘v8::FunctionTemplate::GetFunction()’
npm ERR!   128 |   target->Set(Nan::New("BluetoothHciSocket").ToLocalChecked(), tmpl->GetFunction());
npm ERR!       |                                                                ~~~~~~~~~~~~~~~~~^~
npm ERR! In file included from /home/trougnouf/.cache/node-gyp/19.1.0/include/node/v8-function.h:15,
npm ERR!                  from /home/trougnouf/.cache/node-gyp/19.1.0/include/node/v8.h:33,
npm ERR!                  from /home/trougnouf/.cache/node-gyp/19.1.0/include/node/node.h:73,
npm ERR!                  from /home/trougnouf/.cache/node-gyp/19.1.0/include/node/node_buffer.h:25,
npm ERR!                  from ../src/BluetoothHciSocket.cpp:7:
npm ERR! /home/trougnouf/.cache/node-gyp/19.1.0/include/node/v8-template.h:495:46: note: candidate: ‘v8::MaybeLocal<v8::Function> v8::FunctionTemplate::GetFunction(v8::Local<v8::Context>)’
npm ERR!   495 |   V8_WARN_UNUSED_RESULT MaybeLocal<Function> GetFunction(
npm ERR!       |                                              ^~~~~~~~~~~
npm ERR! /home/trougnouf/.cache/node-gyp/19.1.0/include/node/v8-template.h:495:46: note:   candidate expects 1 argument, 0 provided
npm ERR! ../src/BluetoothHciSocket.cpp: In member function ‘void BluetoothHciSocket::poll()’:
npm ERR! ../src/BluetoothHciSocket.cpp:251:22: warning: ‘v8::Local<v8::Value> Nan::MakeCallback(v8::Local<v8::Object>, v8::Local<v8::String>, int, v8::Local<v8::Value>*)’ is deprecated [-Wdeprecated-declarations]
npm ERR!   251 |     Nan::MakeCallback(Nan::New<Object>(this->This), Nan::New("emit").ToLocalChecked(), 2, argv);
npm ERR!       |     ~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
npm ERR! In file included from ../src/BluetoothHciSocket.cpp:8:
npm ERR! ../node_modules/nan/nan.h:1051:46: note: declared here
npm ERR!  1051 |   NAN_DEPRECATED inline v8::Local<v8::Value> MakeCallback(
npm ERR!       |                                              ^~~~~~~~~~~~
npm ERR! ../src/BluetoothHciSocket.cpp: In member function ‘void BluetoothHciSocket::emitErrnoError()’:
npm ERR! ../src/BluetoothHciSocket.cpp:269:74: error: no matching function for call to ‘v8::Object::Get(v8::Local<v8::String>)’
npm ERR!   269 |   Local<Function> errorConstructor = Local<Function>::Cast(globalObj->Get(Nan::New("Error").ToLocalChecked()));
npm ERR!       |                                                            ~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
npm ERR! In file included from /home/trougnouf/.cache/node-gyp/19.1.0/include/node/v8-array-buffer.h:13,
[... 307 lines cut ...]
113 error make: *** [binding.target.mk:109: Release/obj.target/binding/src/BluetoothHciSocket.o] Error 1
113 error gyp ERR! build error 
113 error gyp ERR! stack Error: `make` failed with exit code: 2
113 error gyp ERR! stack     at ChildProcess.onExit (/usr/lib/node_modules/node-gyp/lib/build.js:203:23)
113 error gyp ERR! stack     at ChildProcess.emit (node:events:513:28)
113 error gyp ERR! stack     at Process.ChildProcess._handle.onexit (node:internal/child_process:293:12)
113 error gyp ERR! System Linux 6.0.9-zen1-1-zen
113 error gyp ERR! command "/usr/bin/node" "/usr/lib/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
113 error gyp ERR! cwd /home/trougnouf/.local/lib/node_modules/bluetooth-hci-socket
113 error gyp ERR! node -v v16.18.0
113 error gyp ERR! node-gyp -v v9.3.0
113 error gyp ERR! not ok
114 verbose exit 1
115 timing npm Completed in 1696ms
116 verbose unfinished npm timer reify 1669125802363
117 verbose unfinished npm timer reify:build 1669125802609
118 verbose unfinished npm timer build 1669125802609
119 verbose unfinished npm timer build:deps 1669125802609
120 verbose unfinished npm timer build:run:install 1669125802613
121 verbose unfinished npm timer build:run:install:node_modules/bluetooth-hci-socket 1669125802613
122 verbose code 1
123 error A complete log of this run can be found in:
123 error     /home/trougnouf/.npm/_logs/2022-11-22T14_03_22_288Z-debug-0.log
```
</details>

krucios, 2023-08-08:
Same issue here with node-gyp v9.3.1
Anybody knows how to approach this issue?

Muhamad-21, 2023-11-28:
Same issue here with node v12.22.9

krucios, 2023-12-01:
I ended up using
```
    "@abandonware/bluetooth-hci-socket": "^0.5.3-10",
    "@abandonware/noble": "^1.9.2-23",
````